tuya-cloudcutter/tuya-cloudcutter #572: Tuya Smart Life Garage Door Motor Receiver 4CH Switch Controller WiFi Switch

Liionboy, 2023-12-13:
Add new profile from this Smart Switch Garage Door:

{
    "timer_arr": {
        "lastFetchTime": 0,
        "cnt": 0
    },
    "gw_bi": {
        "uuid": "40b44efc2e82416d",
        "psk_key": "zX8i8yPWhq4iXTKt6qS7dYQf2dICdGdBlBree",
        "auth_key": "FTgUA1ZnLLMbzBQqJnH5jQRs7yoatSVi",
        "ap_ssid": "SmartLife",
        "ap_passwd": null,
        "country_code": "CN",
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false,
        "fac_pin": "y1zhf5yncrvoojmx"
    },
    "user_param_key": {
        "bt1_lv": 0,
        "bt1_pin": 8,
        "bt2_lv": 0,
        "bt2_pin": 9,
        "bt3_lv": 0,
        "bt3_pin": 10,
        "bt4_lv": 0,
        "bt4_pin": 11,
        "ch_cddpid1": 9,
        "ch_cddpid2": 10,
        "ch_cddpid3": 11,
        "ch_cddpid4": 12,
        "ch_dpid1": 1,
        "ch_dpid2": 2,
        "ch_dpid3": 3,
        "ch_dpid4": 4,
        "ch_num": 4,
        "crc": 109,
        "cyc_dpid": 43,
        "inch_dp": 44,
        "init_conf": 38,
        "jv": 100,
        "module": "CB3S",
        "net_trig": 4,
        "net_type": 0,
        "netled1_lv": 0,
        "netled1_pin": 22,
        "netn_led": 0,
        "nety_led": 0,
        "on_off_cnt": 10,
        "onoff_clear_t": 6,
        "onoff_rst_m": 0,
        "onoff_rst_type": 2,
        "onoff_type": 2,
        "rand_dpid": 42,
        "remote_add_dp": 49,
        "remote_io": 20,
        "remote_list_dp": 50,
        "remote_select": 1,
        "reset_t": 5,
        "rf_width": 345,
        "rl1_lv": 1,
        "rl1_pin": 24,
        "rl2_lv": 1,
        "rl2_pin": 6,
        "rl3_lv": 1,
        "rl3_pin": 26,
        "rl4_lv": 1,
        "rl4_pin": 14,
        "series_ctrl": 0,
        "total_bt_lv": 0,
        "total_bt_pin": 7,
        "total_stat": 0,
        "zero_select": 0
    },
    "gw_di": {
        "abi": 0,
        "id": "bf01d5a35552da09damwv9",
        "swv": "1.3.10",
        "bv": "40.00",
        "pv": "2.2",
        "lpv": "3.4",
        "pk": "keyjnuy4s3kre7m7",
        "firmk": "keyjnuy4s3kre7m7",
        "cadv": "1.0.5",
        "cdv": "1.0.0",
        "dev_swv": "1.3.10",
        "s_id": "000004g6t2",
        "dtp": 0,
        "sync": 0,
        "attr_num": 1,
        "mst_tp_0": 100,
        "mst_ver_0": "100",
        "mst_md5_0": "50a24f9e17f988ea6782bc2b95768961",
        "mst_tp_1": 0,
        "mst_ver_1": null,
        "mst_md5_1": null,
        "mst_tp_2": 0,
        "mst_ver_2": null,
        "mst_md5_2": null,
        "mst_tp_3": 0,
        "mst_ver_3": null,
        "mst_md5_3": null
    },
    "inch_time_key": "HEX:000001020001040001060001",
    "wd_lock_mem": {
        "ctrl_lock_cfg": "AAAAAAAAAAA="
    },
    "gw_wsm": {
        "nc_tp": 9,
        "ssid": "U21hcnQ=",
        "passwd": "bWF5YTIwMjE=",
        "md": 0,
        "random": 0,
        "wfb64": 1,
        "stat": 2,
        "token": "Nmu4gUzD",
        "region": "EU",
        "reg_key": "1DWt",
        "dns_prio": 0
    },
    "tls_ca_cnt": 0,
    "wf_start_md": 3,
    "gw_ai": {
        "key": "S^1hu!hj^R4H5(N}",
        "lckey": "B|SA@'EHEgBbkO=2",
        "h_url": "http://a.tuyaeu.com/d.json",
        "h_ip": "18.156.111.21",
        "hs_url": null,
        "hs_ip": null,
        "hs_psk": "https://a3.tuyaeu.com/d.json",
        "hs_psk_ip": "18.185.182.159",
        "mqs_url": null,
        "mqs_ip": null,
        "mq_url": "m2.tuyaeu.com:1883",
        "mq_ip": "35.156.44.172",
        "ai_sp": null,
        "ai_sp_ip": null,
        "mq_psk": "m2.tuyaeu.com:8886",
        "mq_psk_ip": "35.156.44.172",
        "lp_url": "baal.tuyaeu.com",
        "lp_ip": "18.193.97.90",
        "time_z": "+02:00",
        "s_time_z": "[[1711846800,1729990800],[1743296400,1761440400]]",
        "wx_app_id": null,
        "wx_uuid": null,
        "dy_tls_m": 2,
        "cloud_cap": 1025,
        "psk21_key": null
    },
    "000004g6t2": [
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 1
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 2
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 3
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "bool"
            },
            "id": 4
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 9
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 10
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 11
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "min": 0,
                "max": 86400,
                "scale": 0,
                "step": 1,
                "type": "value"
            },
            "id": 12
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "range": [
                    "off",
                    "on",
                    "memory"
                ],
                "type": "enum"
            },
            "id": 38
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 42
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 43
        },
        {
            "mode": "rw",
            "property": {
                "type": "string",
                "maxlen": 255
            },
            "id": 44,
            "type": "obj"
        },
        {
            "type": "obj",
            "mode": "rw",
            "property": {
                "range": [
                    "flip",
                    "sync",
                    "button"
                ],
                "type": "enum"
            },
            "id": 47
        },
        {
            "mode": "rw",
            "id": 48,
            "type": "raw"
        },
        {
            "mode": "rw",
            "id": 49,
            "type": "raw"
        },
        {
            "mode": "rw",
            "id": 50,
            "type": "raw"
        }
    ],
    "ble_beaconkey": "980C0698E8B3110DC457B819F2B81DAE",
    "astro_timer": {
        "timestamp": 1702492126,
        "index": 0
    },
    "is_stride": 0,
    "wd_cfg_mem": {
        "power_cfg": 0,
        "key_type_cfg": 1
    },
    "mf_test_close": true,
    "em_sys_env": "bk7231n"
}

And i will attach 2 MB dump extract:

[cb3s_bk72xx_2023-12-13_20-28-32.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13665355/cb3s_bk72xx_2023-12-13_20-28-32.zip)

Cossid, 2023-12-13:
This is switch firmware 1.3.10, which is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list and will require a serial flash.

closed 2023-12-13 by Cossid

Liionboy, 2023-12-13:
> This is switch firmware 1.3.10, which is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list and will require a serial flash.

Yes, i put esphome via UART, sorry i didn't know this but i open a topic on Forum with all yaml for esphome and 2 mb dump