tuya-cloudcutter/tuya-cloudcutter #573: "IR8FunctionLightString" Fairy Lights exploit fails

patagonaa, 2023-12-17:
I've had issues getting the exploit to work on some very cheap "smart" fairy lights I found out run Tuya.
On the PCB there's a `BK7231N`, an IR receiver and a transistor half-bridge to switch the LEDs between the two colors. Firmware version shown in the "Smart Life" App is `1.0.4`.
Not much can be found about the product online except the product video linked on the packaging:
http://lighttrade.net/kofur/smarte_lichterkette/

The "proof-of-concept" script seems to work (causes the controller to hang and reboot after some time), however all 1.0.4 profiles I've tried do not seem to work (`Exploit run, saved device config too` but `The profile you selected did not result in a successful exploit` after reboot).

Dumping and flashing via serial (pads on the back of the PCB) works fine: [readResult_BK7231N_QIO_Lichterkette_2023-16-12-18-49-13.bin.gz](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13694400/readResult_BK7231N_QIO_Lichterkette_2023-16-12-18-49-13.bin.gz)

[image]
[image]

Cossid, 2023-12-17:
Added as `Tuya Generic IR 8 Function Fairy String Lights`

closed 2023-12-17 by Cossid

patagonaa, 2023-12-17:
The exploit+flashing worked great with the new profile, thank you very much ❤️