tuya-cloudcutter/tuya-cloudcutter #574: KHSUIN WiFi Smart Light Bulbs (150W) 16W Dimmable 1600LM BR30, E26 RGBCW

59zCez, 2023-12-18:
Hi, I got 4 Bulbs that are KHSUIN 16W, they are the BR30, they look a little different that the ones from the OP, I have tried to load the firmware with no success, see attached log from couldcutter - it seems like it should work and it does not... I have opened one of the bulbs to verify the chipset, indeed is BK7231T.

I saw this refereeing to the round bulbs and the small 6 watt bulbs but thes not seem to work with the BR30
https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/commit/9df4aff4069861b7f540a8871111891fcbc902e7#diff-6b51f8f934f19b2c4be16fbd2346b4a6600998509dfe6123b2ffd20816283883

Product:
https://www.amazon.com/Smart-Light-Bulbs-Equivalent-Changing/dp/B08D3LCBMN/ref=sr_1_1_sspa?crid=19WVQZ5UPUEX4&keywords=KHSUIN%2BBR30%2B16w&qid=1701368848&s=hi&sprefix=khsuin%2Bbr30%2B16w%2Ctools%2C81&sr=1-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

I have loaded one using local tuya and got some useful information that may help (specifications,debuging)... not sure if there is anything missing from the profile already listed/loaded under couldcutter:
Tried to load OpenBeken-v1.17.262_bk7231t.ug.bin and ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin and no success.
I dont own a android equipment so I cant try a dump., is there anything that can be done?
I was able to make it work using TuyaLocal but I really preferred either ESPhome or Tasmota.

Tuya developer shows this as the Product:

Device Name Product
Smart Bulb 1 bk_lm96xe1_16w

{
"result": {
"active_time": 1702902546,
"category": "dj",
"create_time": 1702869514,
"custom_name": "Smart Bulb 1",
"icon": "smart/icon/001458092465966zkPdc/710695a223fb43108cfb908c49b6587a.png",
"id": "xxxxxxxxxxxxxx",
"ip": "96.255.49.102",
"is_online": true,
"lat": "",
"local_key": "xxxxxx",
"lon": "",
"model": "",
"name": "Smart Bulb",
"product_id": "fslmzu1jritagk4z",
"product_name": "bk_lm96xe1_16w",
"sub": false,
"time_zone": "-05:00",
"update_time": 1702902546,
"uuid": "xxxxxxxxx"
},
"success": true,
"t": 1702913369714,
"tid": "xxxxxxxxxxxxxxxxx"

[Debuging from tuya developer.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708585/Debuging.from.tuya.developer.txt)
[Standard Status set.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708586/Standard.Status.set.txt)
[specifications.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708587/specifications.txt)
[KHSUIN 16W.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708588/KHSUIN.16W.txt)

Is there anything that can be done? thanks in advance.

59zCez, 2023-12-18:
Also saw this post refereeing to KHSUIN 16W regular bulb.
https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/172

Cossid, 2023-12-18:
What version of firmware is on the bulbs you have?

If the existing profiles have not worked and the version matched, we would need a full dump to support the device.

59zCez, 2023-12-18:
Hi, SmartLife reads as 1.0.0 firmware.
I loaded the dump software (LightLeak) on an amazon fire tablet (closes thing I can get) and it runs but it does not seem to work... I'll see if I can get something else... I'll report back.
Thanks!

Cossid, 2023-12-22:
Lightleak also requires a devboard to run along side it, it will not work without that portion.

59zCez, 2023-12-22:
Hi, as a devboard do you mean something like A FTDI adapter to USB? I have used it to flash some Sonoff devices to Tasmota with success, not sure how to do it with this bulb. I haven't been able to get the dup just using Lightleak  and still trying to get a android tablet to use - do you have some instructions available that I can follow using the 
[image]
board?
See attached, this a picture that shows the pin connections on the actual board, under the Tasmota site there is one that is the same bulb but the 13W version instead of the 16W, under the 16W bulb I0o (GPIO0) is missing so I could not continue that route, unless one of the 2 above RXD connection is the one to use.
the one 13W:
Tasmota ](https://templates.blakadder.com/khsuin_BL96E.html)

Cossid, 2023-12-22:
No, a dev board as in a ESP8266, ESP32, or beken board you have serial access to and can install the platformio bin included in the LightLeak directions.  There is no guide for it, and if you're asking this basic question, the answer is you probably don't have the requirements.

A Uart board would be for directly connect to the device without LightLeak.  Yes, everything you need is in that picture, RXD1, TXD1, 3.3V, GND, and CEN are the pins you need to serial dump.  There is no IO0 for Beken.  You hook RDX1<->Uart TX, TXD1 <-> Uart RX, 3.3V <-> 3.3V, GND <-> GND, and then have an extra ground for tapping CEN to reset the device, LTChiptool gives instructions for dumping when you run the process.

59zCez, 2023-12-22:
Thank for the explanation, from the picture can you tell which is the CEN pin?

Cossid, 2023-12-22:
There is a pin clearly labeled CEN, to the left of GND

59zCez, 2023-12-22:
Thanks, I will follow your suggestions... I'll report back.

59zCez, 2023-12-23:
Hi again, I was able to get the dump using LTChiptool and got the .bin file (attached), I hope this is all is needed, let me know.
Thanks!

[image]
[ltchiptool_dump_2023-12-23_13-07-00.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13760019/ltchiptool_dump_2023-12-23_13-07-00.zip)

59zCez, 2023-12-23:
The CEN+GND reset combination was bit tricky, it took me a few times to actually get it at the right time so it can start dumping...

Cossid, 2023-12-23:
Yes, that is what we needed.  I have added a new profile as `KHSUIN BR30 16W RGBCT Bulb` (might take a couple minutes to show up).  I have also added A19 to the previous profile, as they are distinctly different.

closed 2023-12-23 by Cossid

59zCez, 2023-12-23:
Nice.. I will give it a try in a few, give it a few minutes to show up and I will report back.. thanks much!

59zCez, 2023-12-24:
Hi, just reporting back, these worked great... I have upgraded firmware on 4 bulbs and added them to ESPHome/Home Assistant without issue, they work great changing colors, etc.!! I lost the default 'Effects' section seen when used with Tuya app or TuyaLocal (that shows under the color scheme) but this can be added by editing the .yaml file and adding the proper entries, this is not so important... now I can use these bulbs as intended.
Thanks so much, I really appreciated.
[image]

Drizzt321, 2024-01-21:
So I'm trying to use ESPHome LibreTiny, and trying to get the right pins doing the right thing. I got the board `generic-bk7231t-qfn32-tuya`, RGB pins configured correctly, but trying to figure out the CCT or Warm White/Cool White LEDs. 

|Color|Pin|
|---|---|
|red|P9|
|blue|PWM5|
|green|PWM4|

PWM2 does seem to control the Cool White, but PWM0 doesn't do anything with Warm White. So not really sure. I'm actually not sure these are tunable white bulbs. 

I only see 1 set of White LEDs, whereas the old version I see both Cool and Warm white LEDs.