tuya-cloudcutter/tuya-cloudcutter #575: Does -f arg require full path? Just bricked 2 devices

Deep-Six, 2023-12-20:
Hello, 
I have had mixed results with the Globe 50329 devices, I managed to get 2 flashed correctly. The successful times I was able to see the esphome kickstarter as an option in the custom firmware screen selection. When I went to flash  2 more tonight the same procedure that worked prior no longer provided the 2nd option only 1 option for firmware. I figured, ok that's strange, so I looked into  how to pass the custom firmware directly and used the following command. sudo ./tuya-cloudcutter.sh -w wlan0 -f ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin Everything looked to be going fine, with the MQTT report statements saying it was nearly finished.  The devices never seem to report and the button is no longer functional only the blue led comes on and stays on.  Is it possible to flash these things via serial, or are they now completely bricked?

Can anyone advise why this happened, is there a problem with the command? Does the filename need to be a full path? If so it might be valuable to show that in the -h command.

Thanks for all the hard work...

Cossid, 2023-12-20:
> When I went to flash 2 more tonight the same procedure that worked prior no longer provided the 2nd option only 1 option for firmware

That is currently not possible unless you removed a file, there are two viable firmwares available for every device currently.  Your log in the second post confirms this.

> Everything looked to be going fine, with the MQTT report statements saying it was nearly finished. The devices never seem to report and the button is no longer functional only the blue led comes on and stays on.

When you flash with 3rd party firmware, the device is unconfigured, or configured at the 3rd party firmware defaults, which means the device is not fully functional yet (buttons will not work, it won't connect to your wifi).  This all indicates a successful flash, the device should have reset into an AP mode that you need to connect to and start the process of device configuration or discovery.

> Does the filename need to be a full path? If so it might be valuable to show that in the -h command.

No path is needed, the help file just says filename, which is correct, and if you don't give it what it expects, it will error out early in the process and tell you so.  There are plenty of safeguards around this.

closed 2023-12-20 by Cossid

Deep-Six, 2023-12-20:
@Cossid  I blame sleep you're 100% correct, I plugged this in this morning and the AP popped up.... PEBCAC!