tuya-cloudcutter/tuya-cloudcutter #577: SSL: NO_SHARED_CIPHER

devLeitner, 2023-12-25:
Hello there, this issue is similar to the already closed issue #497 .

I am trying to flash a Hama GU10 5.5W RGB bulb. 
I checked the firmware in the SmartLife App: 1.1.2
Therefore I am using following firmware: 1.1.2 - BK7231T / oem_bk7231s_light_ty

The script fails at the end, where the light is supposed to connect to the cloudcutterflash AP and receive its new firmware. I am getting this:
```
wlx5ce931af3499: interface state UNINITIALIZED->ENABLED
wlx5ce931af3499: AP-ENABLED 
SSL Error on 15 ('127.0.0.1', 39428): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39430): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39438): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('127.0.0.1', 39446): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39454): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39456): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39466): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('127.0.0.1', 39468): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39484): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 39492): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('127.0.0.1', 39506): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
```

Apparently the light can't connect to the AP, so I tried to to the steps like in #497. These were:

* Setting device into pairing-mode (fast-blink) and trying to add it to the AP in SmartLife.
* Setting device into AP-mode (slow-blink), adding it in SmartLife (--> connecting to the AP myself to transmit the credentials). Nothing worked.

The hotspot is running, because if I connect with my phone I see stuff like following:

```
404 GET /generate_204 (10.42.42.34) 1.18ms
404 GET /gen_204 (10.42.42.34) 0.66ms
404 GET /generate_204 (10.42.42.34) 0.68ms

```

I don't know if you need the full logs, but I can send them asap. I just don't want to clutter the thread so others can find help faster :)

What is going wrong, any ideas?

Cossid, 2023-12-25:
The ssl errors are from your phone and are normal when using that rejoin method, but without the full logs and filtered that way its hard to verify.  It looks like the device never finished connecting.  You may want to run with the `-v` flag to see dhcp requests/status

devLeitner, 2023-12-25:
Oh you're fast.

Gonna do.

I have waited around 3-5 min without doing anything and the terminal stays blank. Should iI wait longer?

P.s. My cloudcutterflash AP knows when I try to connect a lightbulb to the AP? That's interesting.

Gonna post a full log with the option enabled soon.

Thanks for ya fast response

Cossid, 2023-12-25:
Need to see the request line paired with the ssl errors to be sure, but the light will start making specific requests the phone does not when connected.

Cossid, 2023-12-25:
`127.0.0.1` isn't normal for ssl errors, that means something on the device is possibly polling the webserver as well, but there are no requests paired, which is odd.

devLeitner, 2023-12-25:
Running following command:
'''
sudo ./tuya-cloudcutter.sh -w wlx5ce931af3499 -r -v
'''
Following order:

* Clicking through interactive stuff
* Letting AP run for ~5 Minutes doing nothing
* Bringing bulb into AP-mode
* !!! Connecting phone to cloudcutterflash, because SmartLife only allowed me to select the network I was currently connected with
* Connecting bulb to cloudcutterflash
* Joining bulb-network (A-xxxx SSID)
* Bulb turned into emitting continous light again (around 22:08:06 timestamp)
* Letting it run another ~5 min

There:
```
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_light_ty
   1.1.2 - BK7231N / oem_bk7231n_control_switch
   1.1.2 - BK7231N / oem_bk7231n_dimmer_kjd
   1.1.2 - BK7231T / oem_bk7231s_control_switch
   1.1.2 - BK7231T / oem_bk7231s_dltj_2switch_alerc
   1.1.2 - BK7231T / oem_bk7231s_hailai_1_plug
   1.1.2 - BK7231T / oem_bk7231s_light_db
 > 1.1.2 - BK7231T / oem_bk7231s_light_ty
   1.1.2 - BK7231T / oem_bk7231s_rnd_switch
   1.1.2 - BK7231T / oem_bk7231s_ty_ffc_db_ldv
   1.1.2 - BK7231T / oem_bk7231t_light3_laser_nanxin
   1.1.3 - BK7231N / bk7231n-unknown
   1.1.3 - BK7231T / oem_bk7231s_2_plug_arlec
   1.1.3 - BK7231T / oem_bk7231s_double_switch_rmd

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 146631.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231t.ug.bin

Selected Device Slug: tuya-generic-e27-a60-rgbcww-v1
Selected Profile: oem-bk7231s-light-ty-1.1.2-sdk-2.0.0-30.06
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
[... 526 lines cut ...]
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 client provides name: wlan0
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 DHCPREQUEST(wlx5ce931af3499) 10.42.42.33 70:89:76:39:d4:8c 
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 tags: wlx5ce931af3499
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 DHCPACK(wlx5ce931af3499) 10.42.42.33 70:89:76:39:d4:8c wlan0
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 next server: 10.42.42.1
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  1 option: 53 message-type  5
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option: 51 lease-time  12h
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option: 58 T1  6h
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option: 59 T2  10h30m
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option:  1 netmask  255.255.255.0
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option:  3 router  10.42.42.1
Dec 25 22:13:21 dnsmasq-dhcp[16]: 800397794 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 25 22:15:48 dnsmasq[16]: query[A] a3.tuyaeu.com from 10.42.42.33
Dec 25 22:15:48 dnsmasq[16]: config a3.tuyaeu.com is 10.42.42.1
Dec 25 22:15:48 dnsmasq[16]: query[A] a3.tuyaeu.com from 10.42.42.33
Dec 25 22:15:48 dnsmasq[16]: config a3.tuyaeu.com is 10.42.42.1
Dec 25 22:15:49 dnsmasq[16]: query[A] a3.tuyaeu.com from 10.42.42.33
Dec 25 22:15:49 dnsmasq[16]: config a3.tuyaeu.com is 10.42.42.1
Dec 25 22:15:51 dnsmasq[16]: query[A] a3.tuyaeu.com from 10.42.42.33
Dec 25 22:15:51 dnsmasq[16]: config a3.tuyaeu.com is 10.42.42.1

```

Cossid, 2023-12-25:
No idea what is going on here, appears something isnt getting routed correctly.  Something on localhost is connecting to local port 80 without making an http request.  The tuya device connects and makes dns requests, but nothing after, which is abnormal.  Is there a conflicting 10.42. network?  Not sure what platform this is, doesnt look like a pi

devLeitner, 2023-12-26:
No this is a laptop with the latest lts. Internet over etherner and an TPLink Archer Tu3 Pro Wifi Dongle.

Local Network runs on 192.168.x.x so nothing there. I have my own DNS service which routes some stuff, but this should be outside of matter.

Cossid, 2023-12-26:
> Who routes the stuff? NetworkManager? So my local conf? (But -r should override this right?)

NetworkManager handles device configuration, HostAPD & dnsmasq do the routing.  `-r` is just a flag to reset NetworkManager to make sure profiles don't linger.

> Tuya lamp faulty if it doesn't follow up with requests?

No, it's making requests, it's part of the core SDK, but it's probably not fully connecting properly somehow.

> I have another Ubuntu machine around... The docker container only manages the incoming requests on a webserver and the key exchange/exploit stuff (never read source code, sorry I don't know what's happening). So the underlying OS could be the problem?

Possibly, can't say for certain, but what I'm seeing in your logs is not what we normally see.

> So 10.42.42.33 is my bulb, which got its IP from the DHCP in the docker container (?!) and tries to contact tuya servers? Maybe it couldn't resolve the address?

HostAPD & dnsmasq are hosting, and tornado is emulating a Tuya server.  The logs show it requesting dns and getting a response, but it seems to just keep doing it repeatedly, which is abnormal.

devLeitner, 2023-12-26:
Any points where to start? I can try to look into this and find something interesting

Cossid, 2023-12-26:
Not really, no.

devLeitner, 2023-12-26:
The fun begins then, gonna reply if I get it working.

RyantHults, 2024-06-20:
@devLeitner did you ever find out how to resolve this? Currently trying to flash the ESP kickstart firmware on to a Tuya Generic Temperature and Humidity Sensor v1.1.11. My logs look similar to yours, except mine never gets out of the SSL error loop
```
sudo ./tuya-cloudcutter.sh 
Building cloudcutter docker image
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
[1/3] STEP 1/2: FROM python:3.9.18-slim-bullseye AS base
[1/3] STEP 2/2: RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
--> Using cache 2f8d86f211e8ad2a3da2c306daa91f8e46a5e2a1d4a9d3afb3e17dee1b382d2c
--> 2f8d86f211e
[2/3] STEP 1/5: FROM 2f8d86f211e8ad2a3da2c306daa91f8e46a5e2a1d4a9d3afb3e17dee1b382d2c AS python-deps
[2/3] STEP 2/5: RUN pip install --upgrade pipenv
--> Using cache 4edd879f51c93b98996fc0fac8814f2a6d866f028dade18404007570d3751626
--> 4edd879f51c
[2/3] STEP 3/5: COPY src/Pipfile /src/
--> Using cache d3b1aa3e7e5d200933b9b79cd108f2d8ae07c2f8c573f4201a863e50e6bc5997
--> d3b1aa3e7e5
[2/3] STEP 4/5: COPY src/Pipfile.lock /src/
--> Using cache d7d2e1f1e431caca4d0ef7c2a6a282f4293d1b5d8a3642e8eefaaccd8fc4457c
--> d7d2e1f1e43
[2/3] STEP 5/5: RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
--> Using cache f09b7391ddba74d52526360dbc973e42ae4254dc3490934c526d57b42648168d
--> f09b7391ddb
[3/3] STEP 1/3: FROM f09b7391ddba74d52526360dbc973e42ae4254dc3490934c526d57b42648168d AS cloudcutter
[3/3] STEP 2/3: COPY src /src
--> Using cache ef0d42a9437af2363441b4d0845f80dd95a81387704a2b49450505d82ae11372
--> ef0d42a9437
[3/3] STEP 3/3: WORKDIR /src
--> Using cache 1fe593cf55e3ce87f300e1cbfc28bfa51ee7d244120aee43b8ecce7ed6733788
[3/3] COMMIT cloudcutter
--> 1fe593cf55e
Successfully tagged localhost/cloudcutter:latest
1fe593cf55e3ce87f300e1cbfc28bfa51ee7d244120aee43b8ecce7ed6733788
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
[?] How do you want to choose the device?: From device-profiles (i.e. custom profile) manufacturer/device name
   By manufacturer/device name
   By firmware version and name
 ► From device-profiles (i.e. custom profile)

[?] Select device profile: tuya-generic-temperature-and-humidity-sensor-v1.1.11
 ► tuya-generic-temperature-and-humidity-sensor-v1.1.11

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 11518.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service

Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.

Safety checks complete.

Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.262_bk723[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.262_bk723[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.262_bk723[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin08.29_bk7231n_app.ota.ug.bin
 ► ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231n.ug.bin

Selected Device Slug: tuya-generic-temperature-and-humidity-sensor-v1.1.11
Selected Profile: oem-bk7231n-temp-hum-sensor-1.1.11-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
[... 209 lines cut ...]
SSL Error on 15 ('127.0.0.1', 42028): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('127.0.0.1', 42044): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 42054): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('127.0.0.1', 42058): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 17 ('127.0.0.1', 42070): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 18 ('127.0.0.1', 42086): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 19 ('127.0.0.1', 42102): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 35384): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 35400): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 35410): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 53616): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('127.0.0.1', 53632): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 17 ('127.0.0.1', 53644): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 18 ('127.0.0.1', 53656): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
[... cut ...]
SSL Error on 16 ('127.0.0.1', 53696): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 17 ('127.0.0.1', 53710): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 18 ('127.0.0.1', 53716): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 35742): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 35756): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 15 ('127.0.0.1', 35762): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
...
```

devLeitner, 2024-06-20:
No, I didn't get any clean run-througs sadly :(

closed 2025-01-09 by Cossid