tuya-cloudcutter/tuya-cloudcutter #578: Profile for Zen-2c5w

hackintoshlover, 2023-12-28:
Hi,
Merry Christmas.
Thanks for the amazing software.
Attached is the binary dump for Zen-2c5w. (Wifi+RF)
This device has two relays and built-in 433mhz RF link.
The RF only works with their proprietary kinetic switches.
RC-switch doesn't even find them!
[ltchiptool_dump_2023-12-20_23-13-00.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13783739/ltchiptool_dump_2023-12-20_23-13-00.zip)
[https://www.primezen.in/products/zen-2c5w/](url)

Cossid, 2023-12-28:
Added as `Primezen ZEN-2C5W Dual Channel Controller`

closed 2023-12-28 by Cossid

hackintoshlover, 2023-12-28:
Hey,two questions.
1) where is device profile added.
2) How were you able to do it successfully, i tried again by deleting all the files as you said.
Getting the error below:
~/tuya-cloudcutter/profile-building $ python build_profile.py ltchiptool-bk72xx_2023-12-20-rom.bin
[+] Processing file='ltchiptool-bk72xx_2023-12-20-rom.bin' as ltchiptool-bk72xx_2023-12-20-rom
Missing app RBL container. Using a scan pattern instead
Traceback (most recent call last):
  File "/home/pi/tuya-cloudcutter/profile-building/build_profile.py", line 44, in <module>
    extract.run(file)
  File "/home/pi/tuya-cloudcutter/profile-building/extract.py", line 50, in run
    bk7231tools.__main__.dissect_dump_file(input)
  File "/home/pi/.local/lib/python3.9/site-packages/bk7231tools/__main__.py", line 271, in dissect_dump_file
    _, code = __scan_pattern_find_payload(
  File "/home/pi/.local/lib/python3.9/site-packages/bk7231tools/__main__.py", line 163, in __scan_pattern_find_payload
    raise ValueError(f"Could not find end of partition for {partition.name}")
ValueError: Could not find end of partition for app
pi@raspberrypi:~/tuya-cloudcutter/profile-building $ python build_profile.py ebilong-zen_2c5w-and-device-description.bin
[+] Processing file='ebilong-zen_2c5w-and-device-description.bin' as ebilong-zen_2c5w-and-device-description
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xea20]
		extracted to /home/pi/tuya-cloudcutter/profile-building/ebilong-zen_2c5w-and-device-description
	0x129f0a: app - [encoding_algorithm=NONE, size=0xeb1a0]
		extracted to /home/pi/tuya-cloudcutter/profile-building/ebilong-zen_2c5w-and-device-description
Storage partition:
!!! Couldn't unpack KVStorage, perhaps dump file is invalid?
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/bk7231tools/__main__.py", line 307, in dissect_dump_file
    kvs = KVStorage.unpack(storage_data, aes=aes)
  File "/home/pi/.local/lib/python3.9/site-packages/datastruct/main.py", line 455, in unpack
    raise e
  File "/home/pi/.local/lib/python3.9/site-packages/datastruct/main.py", line 446, in unpack
    value = cls._read_field(ctx, field, meta)
  File "/home/pi/.local/lib/python3.9/site-packages/datastruct/main.py", line 324, in _read_field
    if evaluate(ctx, meta.when) is False:
  File "/home/pi/.local/lib/python3.9/site-packages/datastruct/utils/context.py", line 13, in evaluate
    value = v(ctx)
  File "/home/pi/.local/lib/python3.9/site-packages/bk7231tools/analysis/kvstorage.py", line 211, in <lambda>
    when=lambda ctx: ctx.G.tell() < ctx._data_end,
TypeError: '<' not supported between instances of 'int' and 'staticmethod'; while unpacking 'KVStorage.data_blocks'
App code `user_param_key`:
	- not found!
[+] Searching for known exploit patterns
[+] Matched pattern for BK7231N version SDK 2.3.1, payload type ssid
[+] Searching for ssid payload address
[+] ssid payload address gadget (THUMB): 0xBEF51
[+] Searching for finish address
[+] Finish address gadget (THUMB): 0x1B5B7
[!] Storage file not found
[+] SDK: 2.3.1
[+] Device class: bk7231n_common_user_config_ty
[+] Version: 2.0.2
[+] bv: 40.00
[+] firmware_key: key83r5jq9qqeaxt
[!] uuid was not found
[!] Unable to build complete profile as schema remains missing.

I have multiple devices whose profile needs to be built and i don't want to keep disturbing you for the same.

Cossid, 2023-12-28:
Profile is included in the API, it will just show up in the list when you run CloudCutter.

Not sure what that error is.  Did you do `pip install .` from the `profile-building` directory?

hackintoshlover, 2023-12-28:
Yes i did exactly that

hackintoshlover, 2023-12-28:
I have one doubt.
The json file that is being generated by build_profile is already available with ltchiptool.
Meaning when i load my firmware in ltchiptool, it gives me the json file and then identifies that i have a tuyamcu and needs me to download schema.
But it is not able to get the required schema off the internet?
Attached image below for your refrence.
[image]

Cossid, 2023-12-28:
ltchiptool very clearly did download the schema, that's why you see the breakdown.  LTChiptool is trying to process the schema into an ESPHome config, profile building is not doing that, it just passes the schema along in the process, mostly for cut mode.  This device has the full schema in software, it doesn't even need to download it.

I don't know what is causing the error you are seeing from profile building/bk7231tools.  It's possible you need a newer version of bk7231tools.

hackintoshlover, 2023-12-28:
Yes i totally understand that because i am able to find the tuya datapoints for controlling this  family of  devices directly from tuya developer account.
I literally got the datapoints from there and added to esphome yaml and uploaded the same to this device.
Didn't even break open the device.
Attached new device json for your reference, this is a 6 channel wifi + rf switch.
[sn-ct6c10w.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13788159/sn-ct6c10w.txt)

My only problem is why some of the things have missing mapping as stated in the image above.
Do you know a way of how to fix it?

Also from does the ltchiptool download schemas?

Cossid, 2023-12-28:
Those are just because they are not known how to be converted.  All DPIDs are there, but LTChipTool is just saying it hasn't been programmed to create working ESPHome configs from certain datapoints.  There isn't much guidance I can give you there.

Yes, LTChipTool will download schemas if there is proper storage values it can read that are valid.  You can see the full results in the last tab of UPK2ESPHome.

hackintoshlover, 2023-12-28:
Ok.
That puts me in the right direction.
I'll ask on the ltchiptool and esphome page as well if that makes any sense.
Thanks.