tuya-cloudcutter/tuya-cloudcutter #579: Profile building failed

hackintoshlover, 2023-12-28:
Hi,
I have the following https://www.primezen.in/products/zen-2c5w/ tuya based device.
I am trying to build a profile for the same.
I am using debian since windows is failing.
Python v3.9.2.

According to official profile building documentation, if a profile is not found the setup guides through retrieving one from tuya api.
But in my case it just stops.
I made a backup of my firmware using ltchiptool latest version using usb to serial converter.

Kindly help.

Below is my log:
~/tuya-cloudcutter/profile-building $ python build_profile.py ebilong-zen_2c5w-and-device-description.bin
[+] Processing file='ebilong-zen_2c5w-and-device-description.bin' as ebilong-zen_2c5w-and-device-description
[+] Encrypted bin has already been extracted
[+] Haxomatic has already been run
[!] Storage file not found
[+] SDK: 2.3.1
[+] Device class: bk7231n_common_user_config_ty
[!] uuid was not found
[!] Unable to build complete profile as schema remains missing.

Cossid, 2023-12-28:
That log isn't from a first run, so I can't see what the actual issue is, as subsequent runs skip steps if they can.
The bin may be incomplete, or there was an error extracting storage.  Please post the bin here for further analysis.

hackintoshlover, 2023-12-28:
There is no log actually.
As soon as i run profile build command, it outputs the above and exits.
Also when i run profile build command, it says crytptodome not found.
So i installed pycryptodomex after which the command executes as stated above.
Firmware attached in #578

Cossid, 2023-12-28:
That's because it already ran some of it and doesn't re-run it.  It's skipping the extraction step because it has already created a directory, and there is likely an important message in the extraction step.

Cossid, 2023-12-28:
When it fails like that, you must remove the directory it creates and run it again so it starts from the beginning.  I'll add the profile from #578 shortly.

closed 2023-12-28 by Cossid

hackintoshlover, 2023-12-28:
ok
I'll just restart and get back to you.