tuya-cloudcutter/tuya-cloudcutter #580: Not able to set custom 22-char device id

szupi-ipuzs, 2023-12-28:
I have succesfully cloudcut my smart socket (no flashing yet, just cutting), and the cloudcutter set some random 20-chars deviceid.
I am now trying to use it with tinytuya and tuya-local, but they both fail to communicate with the device and I believe I know why:
Originally (ie. before the cut) the deviceid was 22-chars and I AFAIK tinytuya uses this count to decide which protocol it should use.
So, now I am trying to set the original deviceid (via -d parameter), but cloudcutter doesn't seem to like it:

```
Using DeviceId bfe4983bae6083f0e0a9hc and LocalKey 3Q1voW28pldTEyPc

usage: cloudcutter exploit_device

       [-h]

       [--output-directory OUTPUT_DIRECTORY]

       [--deviceid DEVICE_ID]

       [--localkey LOCAL_KEY]

       profile

       verbose_output

cloudcutter exploit_device: error: argument --deviceid: invalid check_arg value: 'bfe4983bae6083f0e0a9hc'

Oh no, something went wrong with running the exploit! Try again I guess...
```

Is there a reason why one cannot set 22char deviceIds?

Cossid, 2023-12-28:
The 20 vs 22 length shouldn't be the issue with TinyTuya, as the protocol it falls back to should work fine, that has been pretty well tested.  It's just not an area much research/testing has been done.  You are welcome to attempt to change the length restrictions.  I think the worst that could happen is the exploit doesn't work, in which case it remains at the previous device id.

szupi-ipuzs, 2023-12-28:
Could you point me to the appropriate place in the code where this is defined?

Cossid, 2023-12-28:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/cdc29466cd279bf569dba2d8c3370cd309fa097d/src/cloudcutter/__main__.py#L397

szupi-ipuzs, 2024-01-04:
Just to follow-up on this - I was not able to set the id, but the reason for this was very poor connectivity to the device - I was getting "802.1X supplicant took too long to authenticate" most of the time. This wasn't cloudcutter fault, it was like that before - i think my wifi dongle is dying slowly :).
I eventually was tired of trying and ended up flashing esphome.

closed 2024-01-04 by szupi-ipuzs