tuya-cloudcutter/tuya-cloudcutter #581: Gateway seems to be 192.168.1.1 on generic bulb E27 RGBCW

Be-Virtual, 2023-12-30:
Hi all,
It seems I have a bulb with a default gateway of 192.168.1.1.
It is recognized as for ex BNETA IoT or an ALIC
Scanning for open Tuya SmartLife AP
..
Found access point name: "SmartLife-6DB8", trying to connect...
Device 'wlan0' successfully activated with 'e09a8183-5b5d-4626-ab6c-531e8e250296'.
Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got 192.168.1.1
Wiping NetworkManager configs

Could you please adapt this piece of code 

Check if successfully connected
        # Note, we were previously checking GENERAL.STATE and comparing to != "activated" but that has internationalization problems
        # There does not appear to be a numeric status code we can check
        # This may need updating if Tuya or one of their sub-vendors ever change their AP mode gateway IP

        AP_GATEWAY=$(nmcli -f IP4.GATEWAY con show "${AP_MATCHED_NAME}" | awk -F ' ' '{print $2}')
        if [ "${AP_GATEWAY}" != "192.168.175.1" ] && [ "${AP_GATEWAY}" != "192.168.176.1" ]; then
            if [ "${AP_GATEWAY}" != "" ]; then
                echo "Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got ${AP_GATEWAY}"
            fi
            if [[ "${i}" == "5" ]]; then
                echo "Error, could not connect to SSID."
                return 1
            fi
        else
            AP_CONNECTED_ENDING=${AP_MATCHED_NAME: -5}
            break
        fi

Thanks

Cossid, 2023-12-30:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

Your device likely isn't Beken and cloudcutter probably will not work.  I'm guessing you have older stock ESP device.  Have you physically checked which module is installed?

Be-Virtual, 2023-12-30:
here the details from tuya:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/PHILIP~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/PHILIP~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Device Name | Device ID | Local Key | IP | UUID | Mac | Product ID | Product | Owner ID
-- | -- | -- | -- | -- | -- | -- | -- | --
TBulb_test | bf012d7da7b9e449a36xg3 | 4bac5ef9aae6b7c3 | 192.168.20.11 | 38802606286dcd5fea66 | 28:6d:cd:5f:ea:66 | qoqolwtqzfuhgghq | TC0000_W600_LIGHT5_RGBCW_P1 | 20439812

</body>

</html>

Cossid, 2023-12-30:
Yeah, that is not a Beken device, looks like probably Winner Micro.  CloudCutter will not work with this device.

closed 2023-12-30 by Cossid

Be-Virtual, 2023-12-30:
Thanks for the feed back.
Any solutions for these kind of bulbs?

Cossid, 2023-12-30:
Serial flashing will likely be your only option.  Looks like OpenBeken may support XR809, but I have no experience with that chip.

kuba2k2, 2023-12-30:
Correction: WinnerMicro might be W600

Cossid, 2023-12-30:
> Correction: WinnerMicro might be W600

Nope, I was saying it could be either, we just can't blindly know.  I believe XRadioTech is owned by Winner Micro, despite being a separate chip.  The mac address prefix matches XR809 modules on the Elektroda forum.

Be-Virtual, 2023-12-30:
If you look at the product column in my table your see W600 (TC0000_W600_LIGHT5_RGBCW_P1)

Cossid, 2023-12-30:
Good point, I was only looking at the mac address prefix, which apparently they (WinnerMicro and XRadioTech) share.  Yeah, you likely have a WinnerMicro W600 device then.  Same advice applies, OpenBeken via serial flashing is currently about the only option.

Be-Virtual, 2023-12-30:
Serial flash is not an option, I tried and completely destroyed the bulb trying to open it.

For the moment, will stay on Tuya Local for these.

Thanks