tuya-cloudcutter/tuya-cloudcutter #582: result for AP_GATEWAY contains newlines, causing looping

DT-one, 2024-01-01:
Hi

I am a little new to this, so sorry if my terminology is not 100% correct.

I was flashing some Arlec energy meters using Lubuntu 16.xx and an old 802.11g wifi dongle (8gb thin client machine and stuff I had lying around) and had a issue where on the second stage of the flash, after the exploit has been run but before wifi is put into AP mode, where the cloudcutter script was not detecting the expected IP propperly. I was getting the message:

`Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got

192.168.175.1` 

Note the newlines.

Looking at the script, I can see the check that is being performed in common.sh. I quickly modified line 91 to:
`AP_GATEWAY=$(nmcli -f IP4.GATEWAY con show "${AP_MATCHED_NAME}" | awk -F ' ' '{print $2}' | grep 192\.168\.17[65]\.1)` 
, adding grep 192\.168\.17[65]\.1 to remove newlines and other nonprintable characters. It fixed the issue. Not sure if it is due to an old version of nmcli?

There is probably a more elegant way to achieve this fix, not that experienced using awk and other linux tools.

Cossid, 2024-01-01:
Running with `-r` should resolve this issue as well.

closed 2024-01-01 by Cossid

DT-one, 2024-01-01:
Ah righto.

Well, the script could easily be made more robust to this regardless, running in -r mode is not an obvious thing to try.

DT

Cossid, 2024-01-01:
You're welcome to submit a PR, I haven't had a solid reproduction case to test against, so the FAQ has mostly covered it so far.