tuya-cloudcutter/tuya-cloudcutter #584: Globe 50323 profile didn't work

Deep-Six, 2024-01-01:
Hey'll happy new year!

I've tried the Globe 50323 profile and it didn't work:

Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Globe Electric
   Feconn
   Feit
   Filohome
   Firefly
   Fitop
   Geeni
 > Globe Electric
   Globo Lighting
   Gosund
   Hama
   Helloify
   Hihome
   Hombli

[?] Select the article number of your device: 50323 RGBCT Recessed Light
   35798 A19 RGBCT Bulb v1.3.21
   35804 RGB E26 LED Bulb
   35851 E26 RGB Bulb
   37783 RGBCT Bulb v2.0.0
   37783 RGBCT Bulb v2.0.3
   50151 Smart 2-Outlet Outdoor Plug
   50239 Motion Night Light
 > 50323 RGBCT Recessed Light
   50329 Smart Plug
   50347 Smart-Plug

[?] Select the firmware version and name: 1.1.0 - BK7231T / oem_bk7231s_light5_sleepwake_random_globe
 > 1.1.0 - BK7231T / oem_bk7231s_light5_sleepwake_random_globe

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231t.ug.bin

Selected Device Slug: globe-electric-50323-rgbct-recessed-light
Selected Profile: oem-bk7231s-light5-sleepwake-random-globe-1.1.0-sdk-1.0.2-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP

Found access point name: "Globe Suite-0D56", trying to connect...
Device 'wlan0' successfully activated with '8e2dbefc-7ed7-4025-beee-d716e0d832fa'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/1tTMORohtR3Z.deviceconfig
Saved device config in /work/configured-devices/1tTMORohtR3Z.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.............
Found access point name: "Globe Suite-0D56", trying to connect...
Device 'wlan0' successfully activated with '8e2dbefc-7ed7-4025-beee-d716e0d832fa'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

I'll try to extract the firmware again, as I think this was based on my original submission, unless we're confident it's correct?

Thanks for all your hard work!

Cossid, 2024-01-01:
The profile for 1.1.0 is correct, but it looks like 1.1.0 may have been an update according to [the original issue](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/283) and we don't have the firmware/profile for the firmware version prior.

Your options are to
A) dump the prior firmware and get it supported properly
B) update the firmware in Smart Life to 1.1.0 (if that's the choice it gives you) and cut from there.

Deep-Six, 2024-01-02:
Ill try that and see how it goes, is 1.1.0 the latest firmware? Im worried i might get a different update that might lock me out of cutting entirely?

Cossid, 2024-01-02:
I don't know, but the screen where you normally see what firmware your device currently has will instead show what it would be upgrading to, so you should know before you actually trigger it.

Deep-Six, 2024-01-02:
Worked exactly as you suggested, updated to 1.1.0 and it patched just fine. I only dug out the same harness I had made to test with today after about 10 months it looks like. It was super easy, do you have an esphome config for the light?  The only one I see close is [Globe 4 Inch 9W Downlight 50359](https://devices.esphome.io/devices/Globe-4-Inch-9W-Downlight-50359)[us](https://devices.esphome.io/standards/us)[esp32](https://devices.esphome.io/board/esp32)
but it's esp32 based and not beken based.

Cossid, 2024-01-02:
https://upk.libretiny.eu/?profile=globe-electric-50323-rgbct-recessed-light

closed 2024-01-02 by Cossid

Deep-Six, 2024-01-02:
> Caution: This email originated from outside of the organization. Do not click links, open attachments, or scan QR codes unless you can confirm the sender and know the content is safe. Report suspicious messages using the ‘Report Message>Phishing’ button in Outlook.
>
> https://upk.libretiny.eu/?profile=globe-electric-50323-rgbct-recessed-light

Worked perfectly!