tuya-cloudcutter/tuya-cloudcutter #588: Maually esiting profile / understanding profile

hackintoshlover, 2024-01-07:
Hi,
Referring the picture below. Can you tell me how to retrieve key from tuya clud api?
[image]

What is address finish, address ssid, ssid padding and how to retrive the same?
[image]

Cossid, 2024-01-07:
This is outside the scope of CloudCutter.

CloudCutter includes the list of DPIDs in the schema (the `id` fields), but doesn't determine how to use them.  Beyond that you'll need to use external tools such as LTChipTool's UPK2ESPHome Schema Download using the key from the CloudCutter profile as either the firmware key or product key.

closed 2024-01-07 by Cossid

hackintoshlover, 2024-01-07:
Okay.
What about the adrress finish, address ssid, ssid padding?
Trying to understand it to find a workaround for devices using esp based tuya ic.
I myself happen to have lots of them.
Tuya convert is able to detect them but unable to exploit.

Cossid, 2024-01-07:
> What about the adrress finish, address ssid, ssid padding?

Those are for CloudCutter exploiting, and are not a part of device configuration.