tuya-cloudcutter/tuya-cloudcutter #591: Tuya Generic Temperature and Humidity sensor - Firmware Dump

t0mm0117, 2024-01-12:
Firmware dump from a generic temperature and humidity sensor using ltchiptool. 

Ltchiptool returns the following:
E: UPK: The chosen device doesn't contain pin configuration.

Possible causes:
- it has vendor-specific firmware
- it uses TuyaMCU (report error if that's the case!)

Auto-generating ESPHome YAML is not possible.

It uses a module labled "CBU" and does not have any other distinguishing marks on it.

[ltchiptool_bk72xx_2024-01-12_23-35-31.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13919434/ltchiptool_bk72xx_2024-01-12_23-35-31.zip)

Cossid, 2024-01-12:
Added as `Tuya Generic TH08 Temperature and Humdity Sensor v1.0.0`

The device class shows the model to be TH08, which I'm assuming may only have been on the box, these temperature sensors usually seem to come with no labeling on the device itself.  It does indeed at least include model-specific firmware.  If this is not a TuyaMCU device, there isn't any pin configuration info anywhere, and would need to be probed manually.  We can't directly tell from this one as it doesn't have the tell-tale `user_config_ty` firmware markings.

closed 2024-01-12 by Cossid

divadiow, 2024-03-13:
> which I'm assuming may only have been on the box

correct. undoubtedly just a TH01 variant
[image]
[image]