tuya-cloudcutter/tuya-cloudcutter #592: [New Device] Geenie GNC-WW217 OTA Flashing to ESPHome

yuuzhan, 2024-01-12:
I know #203 says not to post here, however I was unable to gather a .bin (I used OTA to flash)

I dont have a profile, but if you can tell me how to generate one from a flashed item OTA I can do that and do a pull request there.

I will post the info I used to flash them here, so someone can find it in the future (feel free to close this if there is nothing else I can do to help get it in officially).

Main Module Version: 1.1.8
MCU Module Version: 1.1.8

Process to Flash:

1. Start tuya-cloudcutter.sh
2. Select 2 (Flash 3rd Party Firmware)
3. Select the second option: (By firmware version and name)
4. choose "1.1.8 - BK7231N / oem_bk7231n_plug"
5. follow the usual procedures from here.

Cossid, 2024-01-12:
It really doesn't work that way.  After you have used CloudCutter, certain details that are needed to create a new profile specific to the device have already been overridden, which is why we need a full bin before exploit.  The information removed in the cut/exploit is part of what we need to properly build a verified profile.

The only way we can potentially generate a working profile afterwards is if the device was previously activated on Tuya's services and part of that information are still present in storage.  You can pull the storage data from Kickstart with LTChipTool and post that here, and we can review it, but short of that, we just don't add new profiles when they are not verified, by firmware version and name is sufficient enough.

yuuzhan, 2024-01-12:
Kickstart is not on there anymore, i moved it to ESPHome already.  Just wanted the text that this device works searchable from others without a place to put it without a formal profile made.

If I had thought about it earlier I would have grabbed the info.  Might be useful to have a section in the docs about how to extract the bin OTA.  if there is one, I didt see it :(

closed 2024-01-12 by yuuzhan

Cossid, 2024-01-12:
CloudCutter itself does not have a way to extract the bin OTA.  LightLeak exists, but it is incomplete and a has a higher difficulty, so we generally only push it as a last option.