tuya-cloudcutter/tuya-cloudcutter #597: Arlec PC288HA Twin Plug

BenJamesAndo, 2024-01-18:
Full dump from OBK Web App. Chip BK7231T.
Main Module 1.3.5
MCU Module Version: 1.3.5

[BK7231T_TuyaConfig_obkB9DADCD6.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13974960/BK7231T_TuyaConfig_obkB9DADCD6.zip)

Cossid, 2024-01-18:
The zip file attached is only 11KiB resulting in a 72KiB bin file, which is not a valid full dump.  Please re-open if you are able to attach a full 2MiB dump.

closed 2024-01-18 by Cossid

BenJamesAndo, 2024-01-18:
Oops, here's the full 2MB dump.
[BK7231T_QIO_obkB9DADCD6.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13982222/BK7231T_QIO_obkB9DADCD6.zip)

Cossid, 2024-01-18:
Unfortunately, this is a dump after CloudCutter has been run and 3rd party firmware has been installed, meaning we can't easily generate a profile from this.

BenJamesAndo, 2024-01-18:
Gotcha. I'll aim to get the dump earlier on next time. Cheers.

Cossid, 2024-01-18:
I was able to piece enough of it together with an existing bin to get a profile generated, added as `Arlec PC288HA Twin Plug`