tuya-cloudcutter/tuya-cloudcutter #598: Gosund SW5 smart wifi switch, need dump or profile?

mtntrx, 2024-01-19:
My searching capabilities might have come up short, but I'm not finding this one.  Here are the details.  If it needs work (e.g. dump submitted, etc... ) I'm willing to help.  I have successfully serial flashed one, but would like to make the cloud cutter work.

Please let me know if I missed this device/profile in my search, or if I can/should help work on it.

Details
Externals:
gosund SW5 smart wi-fi switch
Model: SW5
Manufacturer: Shenzhen Cuco Smart Technology Co Ltd (from box)
date code 11/23

Internals:
PCB: SW5-A-V2.1
Module: Z2-N-V1.1  has other "0717" marking
MCU: BK7231N

Smart life app reports:
Main Module: V1.0.1
MCU Module: V1.0.1

Cloudcutter fails with these profiles: 
Tuya generic mini smart switch (listing the BK7231N chip)
gosund-sw5-a-v2.1-smart-switch

test_device_exploitable.py gets a reaction:
1) slow blink stops
2) AP stays present... ~50secs?  (not sure)
3) AP disappears, and switch recovers

Cossid, 2024-01-19:
A new dump would be needed.  The dump we currently have for that switch is 1.0.2, and Gosund uses firmware specific to their devices, so they're not shared with generic profiles, meaning we'd need a dump to support this firmware version of the device.

mtntrx, 2024-01-19:
Ok, so it should be exploitable?  I'll work on learning how to dump and get it submitted.  Thanks!

Cossid, 2024-01-19:
No guarantee, but most likely yes.  1.0.1 should be older than 1.0.2 which was still vulnerable.

mtntrx, 2024-01-19:
great... I'll work on it.   Crazy fast response Cossid.  wow.

mtntrx, 2024-01-20:
[bk7321_gosund_SW5_1.0.1.bin.gz](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13996625/bk7321_gosund_SW5_1.0.1.bin.gz)

Created with: 
`$ ltchiptool flash read BK7231N bk7321_gosund_SW5_1.0.1.bin`

mtntrx, 2024-01-20:
[bk7321_gosund_SW5_1.0.1_r2.bin.gz](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13996705/bk7321_gosund_SW5_1.0.1_r2.bin.gz)

created with 
`$ ltchiptool flash read BK7231N`

Cossid, 2024-01-20:
Added as `Gosund SW5 A V2.1 Smart Switch BK7231N v1.0.1`.  Because this is a device-specific firmware, there is not pin configuration present in storage, so you'll need to use pinscan or trace them out.

closed 2024-01-20 by Cossid

mtntrx, 2024-01-23:
Pin config I get.  Should I learn to submit a pin config?  Let me know.

P7: LED (main, green, most visible)
P8: Btn
P14: Rel
P16: LED_n 1 (red, visible through same switch window as green)

I'd still like to find another channel that could be soldered in.  I see headers for RX2 and TX2, but no luck so far.

Cossid, 2024-01-23:
> Pin config I get. Should I learn to submit a pin config? Let me know.

This isn't really a way to do so, device that have a pin config come from Tuya data.  Technically you could generate it for device that don't have it, but there are no tools to do so.

> I'd still like to find another channel that could be soldered in. I see headers for RX2 and TX2, but no luck so far.

TX2 and RX2 are usable pins, but you would need to disable uart log output on whatever firmware you're using to avoid interference.
TX2 is P0, and RX2 is P1

mtntrx, 2024-01-23:
Big thanks for the hint on TX2 and RX2.  Had the scope out and saw stuff other than what I was trying to do... figured it was likely the UART. Back to code hacking :)