tuya-cloudcutter/tuya-cloudcutter #599: Denver SHP-102 giving trouble

flixman, 2024-01-20:
I recently bought a Denver SHP-102, and I am running arch linux without network manager. I have modified slightly your scripts to replace the nmcli commands with echo statements asking to connect to the AP, etc. (when successful, I will share the findings with the project). This is the log I am getting on the command line:

```./tuya-cloudcutter.sh -r -w wlp5s0f4u2u3u3 -p denver-shp-102-plug -f ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
Successfully built docker image
Loading options, please wait...

Performing safety checks to make sure all required ports are available
Make sure the following ports are not in use. As root, execute: 
(ss -lnpH -A udp sport 53 or sport 67 && ss -lnpH -A tcp sport 80 or sport 443 or sport 1883 or sport 8886) | awk '{if(NR>1) print $6}'
Press ENTER when done...

Make sure the firewall is not running. Press ENTER when done...

Make sure the module rtl8192cu is not blacklisted. Press ENTER when done...

Safety checks complete.

Selected Device Slug: denver-shp-102-plug
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Make sure you are connected to the AP of the appliance (SSID ending in -[A-F0-9]{4}), and press ENTER to continue
detect the AP SSID: iw wlp5s0f4u2u3u3 scan | grep -oP 'SSID: \K.*-[A-F0-9]{4}'
connect to the AP: iw dev wlp5s0f4u2u3u3 connect $SSID
confirm we are connected: iw dev wlp5s0f4u2u3u3 link

Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/9jGDsTLLorGH.deviceconfig
Saved device config in /work/configured-devices/9jGDsTLLorGH.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Make sure you are connected to the AP of the appliance (SSID ending in -[A-F0-9]{4}), and press ENTER to continue
detect the AP SSID: iw wlp5s0f4u2u3u3 scan | grep -oP 'SSID: \K.*-[A-F0-9]{4}'
connect to the AP: iw dev wlp5s0f4u2u3u3 connect $SSID
confirm we are connected: iw dev wlp5s0f4u2u3u3 link

================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

if checking the file with the device config, it contains the following:
```json
{"uuid": "68LSjSPeE6AL", "auth_key": "8T0UsUjbVmvpjjxT", "local_key": "HzVJR1mkNaa17jLL", "sec_key": "i1mTeohg4yse1q1t", "device_id": "fVQLCcUOaDP8LVRikNYD", "chip_family": "BK7231N", "profile_name": "1.1.8 - BK7231N / oem_bk7231n_plug", "device_slug": "denver-shp-102-plug"}
```

Every time I do an attempt a new file gets created with a different uuid, local_key, sec_key, device_id, and auth_key. When slow blinking, the light is violet (in case that matters). What might be going wrong?

Cossid, 2024-01-20:
It looks like you are printing the commands that should be used to connect to networks (`iw dev ...`) instead of executing them?  There isn't much more that can be advised without seeing the modifications you've made or the outputs from said commands.

flixman, 2024-01-21:
hey @Cossid , yesterday I spent a bit of time on this and I made it work with few modifications:
* nmcli is not used, just plain iw and ip commands are used.
* I am running it as a regular user so I have the list of commands that require sudo to work (so that this can be enabled only when required in /etc/sudoers.d/<whatever> file
```
<user> ALL=(root) NOPASSWD: /usr/bin/ip *, /usr/bin/iw *, /usr/bin/dhclient *, /usr/bin/ss *, /usr/bin/systemctl *, /usr/bin/kill *, /usr/bin/tail *
```
I am attaching the patch file with the changes I had to introduce to make it work. If you feel like can be a contribution to the project, I can create a pull request for it.
[drop_nmcli_sudo_requirements.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14001757/drop_nmcli_sudo_requirements.txt)

Cossid, 2024-01-21:
I like the idea of dropping nmcli requirements for iw.  This was just originally written with nmcli and nobody has taken the time to look for a more standardized solution.  At some point I'd also like to rip docker out, as it's really just being used as a python environment, and we could probably just manage that in other ways.

You're not really removing sudo requirements, you're just moving them to be pre-approved, which isn't something we'd want users to have to do to run this, so we would leave the sudo requirement in place.  It makes it more clear what the expectation is.

Feel free to create a fork and submit a PR, and I'll review it when I get some time and have devices to test on.

Cossid, 2024-01-21:
Also, we don't want to introduce that "Enter to continue" change, as some devices don't require interaction to get back into AP mode if you wait long enough.  Because the waiting indicator is just a single dot at a time (instead of a whole row), the prompt still remains readable for instruction.

flixman, 2024-01-22:
Got it, will try to get the PR done during this week :-)

flixman, 2024-01-24:
created: https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/602

closed 2024-01-24 by flixman