tuya-cloudcutter/tuya-cloudcutter #600: Thanks.  Great experience on a bk7231n custom flash (not really an issue)

mtntrx, 2024-01-21:
Big thanks to all those that have built this and the open-bk firmware.  I'm totally impressed.  I've been able to integrate new bk7231n based hardware into my custom ESP/toasmota environment with very reasonable effort.

I was pretty quickly able to do what I needed including serial flash OBK, custom build of my own OBK version, and do an OTA tuya-cloudcut with my own firmware.

Super impressed by all this stuff.  Especially @Cossid 's lightning fast responses while creating the new device-profile based on a device that needed a new firmware dump (my HW did not have a current device-profile)

Question:  Along the way I got hung up a few times and I'm thinking of 3-4 error messages and tweaks that could have really sped up my learning of the whole process.  Would the developers welcome that input, prefer a git pull-request, etc?  If so, I'd like to give back, so let me know.  

Thanks again

Cossid, 2024-01-21:
Feedback is always welcome, and messages can/will be adjusted if appropriate.  Sometimes things don't have a clear pass/fail case, so the messages in some cases are intentional as they are.  Feel free to post specific examples and suggestions.

mtntrx, 2024-01-22:
All small nits.  Just things that would have helped.

HOST_SPECIFIC_INSTRUCTIONS.md
* Great step-by-step.   the location of /etc/dhcpcd.conf  might have moved (maybe I got the wrong install) it was at /etc/dhcp/dhcpclient.conf

* get_input.py:  error message when using -p says "profile not found in API" could add the reminder that if you haven't loaded the profile through the interactive method, it will never be available.  That is listed in the README somewhere, but I had forgotten (or not made the connection)

* get_input.py: when using -f and a custom bin file, it might come up as "incompatible" if the cpu text "bk7231n" is not in the filename.  Error message could be augmented to specify which requirement was missed.  It's awesome to have the error checking to reduce bricks, but I wasn't expecting the filename to be the issue.

again... just small nits

closed 2024-01-22 by Cossid