tuya-cloudcutter/tuya-cloudcutter #601: Cannot connect to device AP

grego1981, 2024-01-22:
Hi I have a door sensor and try to flash it using the tuya-generic-cb3s-door-sensor-v1.0.10.json profile. I use the following command to set up the device:
 sudo ./tuya-cloudcutter.sh -d basmnt_door_sensor -s SSID password -p tuya-generic-cb3s-door-sensor-v1.0.10

the script starts and at some point it is at Scanning for open Tuya SmartLife AP where it stands forever... The device is on ap mode and the SSID it broadcasts is SmartLife-2A64. I have tested with the proof of concept script and it worked fine... Can you please help me?

Cossid, 2024-01-23:
> -d basmnt_door_sensor

is not valid.  `-d` must be paired with `-l` or it will be ignored, and the value you gave for `-d` I don't believe is valid.  You should leave those to be generated if you don't need very specific values.

`-p` requires you to have either manually downloaded, or selected the profile once through the device selection menu.

> the script starts and at some point it is at Scanning for open Tuya SmartLife AP where it stands forever... The device is on ap mode and the SSID it broadcasts is SmartLife-2A64. I have tested with the proof of concept script and it worked fine... Can you please help me?

Make sure devices are within range of each other, AP mode is very weak, especially on battery powered devices.  There is no other advice we can give, as if they are in range, they will be detected and attempt to connect.  At that point in the script, it's literally just wifi scanning and connecting, nothing special going on.

grego1981, 2024-01-23:
Hi and thanks for your answer, i tried without any options selecting manually the profile from the list and it has the same behaviour, the distance between the raspberry pi zero 2w is 5 cm still no go. If i connect to the decive ap from my laptop it gets an ip (192.168.175.100) and the gateway responds to ping in 192.168.175.1

Cossid, 2024-01-23:
Have you fully set up NetworkManager per the instructions at https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md#generic-zero-2w-3-or-4 ?

grego1981, 2024-01-24:
Just retraced all my steps... There was an error in /etc/dhcpcd.conf where denyinterfaces wlan0 was written as denyinterface wlan0... Sorry for the childish mistake and thank you all for your help!!

grego1981, 2024-01-24:
Closing as user error

closed 2024-01-24 by grego1981