tuya-cloudcutter/tuya-cloudcutter #603: Tongou TO-Q-SY1-JWT v1.0.18

Bl00d-B0b, 2024-01-24:
Hello,

I need help with new version of Tongou.  in you library you have tongou 1.0.2 which is based on sdk 2.3.1. 

But the new device is v1.0.18 and based on SDK 2.3.3.
I download firmware, and try to build profile. 

RBL containers:
        0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xea20]
                extracted to dump_extract_dir
Missing app RBL container. Using a scan pattern instead
        0x11000: app - [NO RBL, size=0xf0e00]
                extracted to dump_extract_dir
Storage partition:
        0x1ee000: 68 KiB - 14 keys
        - 'gw_bi'
        - 'user_param_key'
        - 'gw_di'
        - 'tls_ca_cnt'
        - 'wf_start_md'
        - 'timer_arr'
        - 'astro_timer'
        - 'gw_wsm'
        - 'gw_ai'
        - 'is_stride'
        - 'mf_test_close'
        - 'ble_beaconkey'
        - '0000047qny'
        - 'em_sys_env'
                extracted all keys to dump_extract_dir/Tongou_to-q-sy1-jwt-din-rail-switch-with-energy-meter_storage.json
Storage area `user_param_key`:
        - found! Extracted to dump_extract_dir/Tongou_to-q-sy1-jwt-din-rail-switch-with-energy-meter_user_param_key.json

but then run:
./tuya-cloudcutter.sh -r

I can't flash. maybe I do something wrong?

Ed

Cossid, 2024-01-24:
You have only extracted storage data, you have not created a cloudcutter profile.

You can either post the firmware dump here and I will generate one (which would make it available to all if it is not patched), or you can generate one using the tools https://github.com/tuya-cloudcutter/tuya-cloudcutter/tree/main/profile-building

Bl00d-B0b, 2024-01-24:
when I try to generate profile, I see an error:
 Directory has not been fully processed, unable to generate classic profile

[tongou-to-q-sy1-jwt-din-rail-switch-with-energy-meter.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14043766/tongou-to-q-sy1-jwt-din-rail-switch-with-energy-meter.zip)

Cossid, 2024-01-24:
```
[+] Searching for known exploit patterns
[+] Matched pattern for BK7231N version SDK 2.3.3 LAN 3.4/CAD 1.0.5, payload type ssid
[!] The binary supplied appears to be patched and no longer vulnerable to the tuya-cloudcutter exploit.
```

Unfortunately, this device is indeed patched.  Serial flashing will be the only option for this device.

Bl00d-B0b, 2024-01-24:
I also have another socket with power meter, an CBU board with the same problem, attaching

[BSEED_WiFi-Smart-Socket-Energy-Monitoring-16A.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14043785/BSEED_WiFi-Smart-Socket-Energy-Monitoring-16A.zip)

purchased from [https://www.aliexpress.com/item/1005003350708224.html](url)

Cossid, 2024-01-24:
> I also have another socket with power meter, an CBU board with the same problem, attaching [...]

This one is firmware 1.1.15 and already on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.

Bl00d-B0b, 2024-01-24:
Thanks,

yes, serial patching works for both... 

Ed

closed 2024-01-24 by Bl00d-B0b