tuya-cloudcutter/tuya-cloudcutter #604: ELEGRP SPN10 Single Pole Switch [UART flashed still no b AP]

thirug010, 2024-01-27:


Cossid, 2024-01-27:
No idea what this means or what you're even looking for.  Uart flashing has nothing to do with CloudCutter.

thirug010, 2024-01-27:
I tried the OTA flash with couple of firmware it does not work, the device original firmware version was 2.2.4 then update to 3.2.4 still OTA is not working, I opened soldered the pins it took the dump with ITCHIPTOOL and flashed with both "OpenBK7231T/OpenBeken" and "esphome-kickstart" flash is success but still not Wifi OpenBK AP. tried both itchip and 'BK7231Flasher' to flash it several time it does not shows the Wifi AP. 

Some Observations:
1. the Device never go into slow blink mode ( tried the button hold and power on of still not go into slow blink)
2. even though it is fast blink still show tuya / smartlife Wifi AP and works with Tuya App just fine 
3. holding the button and powering on puts the device in slow blinking, not sure this AP / EZmode.
4. When I use the itchip tool analyze the firmware dump it show the message as  "- it has vendor-specific firmware, - it uses TuyaMCU (report error if that's the case!)"

Attaching
[ltchiptool_dump_2024-01-27_02-07-12-001.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14073639/ltchiptool_dump_2024-01-27_02-07-12-001.zip)
 (Note: extracted twice to make sure it full and the fixe size looks full).

thirug010, 2024-01-27:
> No idea what this means or what you're even looking for. Uart flashing has nothing to do with CloudCutter.

hit enter created the issue before entering the full information, Please reopen it.

Cossid, 2024-01-27:
I can't really help with your UART flashing issues, as you have no provided enough useful information (which actual files you used, what offsets you flashed to, etc) and that is generally outside the scope of CloudCutter anyway.

I have renamed the existing profile to `ELEGRP SPN10 Single Pole Switch v2.2.4` and added a new one for the provided as `ELEGRP SPN10 Single Pole Switch v2.3.4`

closed 2024-01-27 by Cossid