tuya-cloudcutter/tuya-cloudcutter #606: Feature request: exclude devices on scanning

GijsTimmers, 2024-01-28:
I am living in a dense urban area. I'm trying to flash a Tuya device, but my neighbour's `eufy RoboVac 35C-xxxx` always pops up first in the list, making it hard to flash my device.

Please add a way of excluding devices for flashing.

Cossid, 2024-01-28:
I'm not certain excluding is the right approach, as opposed to including what is intended.  I'll think about what may be the best approach here.

In the mean time, you can modify `common.sh` line 4 from `AP_CONNECTED_ENDING=""` to `AP_CONNECTED_ENDING="-ABCD"` where `-ABCD` is the ending of the SSID you are attempting to exploit.

GijsTimmers, 2024-01-28:
Thank you, that will work for now.

Cossid, 2025-01-09:
The inverse of this was added instead.  You can use `-o SPECIFIC-AP` to lock to a specific AP.

closed 2025-01-09 by Cossid