tuya-cloudcutter/tuya-cloudcutter #607: New device request: LSC Smart Power Plug, version firmware version V2.0.1

GijsTimmers, 2024-01-28:
I am trying to flash this LSC Smart Power Plug, version `970761.1` . `lsc-2578685-970766-smart-plug-cb2s.json` already exists but that one doesn't work on this device, as that one is using V1.1.7.

According to the Tuya app, the current firmware version on this devices is V2.0.1

What other information should I provide?

Cossid, 2024-01-28:
There are currently no profiles for 2.0.1, so we would need a full firmware dump to support this device.

GijsTimmers, 2024-01-28:
[image]

Could be either a Beken device or a Realtek device.

Cossid, 2024-01-28:
A full dump would be required then.  You can either dump via UART with [LTChipTool](https://github.com/libretiny-eu/ltchiptool) or try your luck [LightLeak](https://github.com/tuya-cloudcutter/lightleak) (requires a UART programmable dev board and an android device)

GijsTimmers, 2024-01-29:
OK, I have set up the requirements to run LightLeak yesterday and will try to run it tonight.

Dumping straight from the board will be very hard as I am very bad at soldering.

In the meantime, [I found a teardown on this device](https://www.huizebruin.nl/elektronica/handleidingen/lsc-stopcontact-ombouwen-met-een-esp12/). According to this website, the device contains a WB2S chip, based on ARM Cortex M4.

With this information provided, do you have any other ideas on flashing this device?

Cossid, 2024-01-29:
No, getting a dump will be the only way.  Being that is WB2S which is BK7231T, there is a high chance this is not patched at least.

Cossid, 2024-02-05:
Complete in #615

closed 2024-02-05 by Cossid

GijsTimmers, 2024-02-06:
I don't think this can be closed yet, this issue is about plug 970761.1, #615 is about 970766.

Cossid, 2024-02-06:
Sorry, got models mixed up.  But until a dump is provided, there is no action that can be taken either.  I will re-open if a dump is provided.

Cossid, 2024-02-10:
This is kind of a long shot, but I found an old incomplete dump for a BK7231T plug with firmware version 2.0.1.  I have added a generic `Tuya Generic Plug v2.0.1` profile you are welcome to try, but don't get your hopes up too high.  It was from a Calex plug, but without the missing part, I just based it on other generic Tuya devices.

GijsTimmers, 2024-02-10:
Sadly, I don't own this device anymore, I exchanged it with the flashable `970766`  version (1.1.8). Hopefully, somebody else is able to test this profile.

tobyworks, 2024-10-19:
can confirm i can flash kicktart on here through the generic plug, however after that im lost, im not sure how to make the plug controllable once esphome is on there