tuya-cloudcutter/tuya-cloudcutter #608: Smart Touch Switch 8 Gang

rt400, 2024-01-28:
Hi.
I have this module
https://www.aliexpress.com/item/4000297406413.html?spm=a2g0o.order_list.order_list_main.83.29a91802dNow78
[image]

Here a Full Dump 
[readResult_BK7231N_QIO_2024-28-1-18-44-34.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14077382/readResult_BK7231N_QIO_2024-28-1-18-44-34.zip)

Thanks for all the help...

Cossid, 2024-01-28:
Added as `Tuya Generic 8 Button Touch Switch`.  There is no device configuration as this is a TuyaMCU device.

closed 2024-01-28 by Cossid

brunohoff, 2024-02-11:
I was able to make all buttons work with the following autoexec.bat

`startDriver TuyaMCU

tuyaMcu_defWiFiState 4

setChannelType 1 toggle
linkTuyaMCUOutputToChannel 1 1 1

setChannelType 2 toggle
linkTuyaMCUOutputToChannel 2 1 2

setChannelType 3 toggle
linkTuyaMCUOutputToChannel 3 1 3

setChannelType 4 toggle
linkTuyaMCUOutputToChannel 4 1 4

setChannelType 5 toggle
linkTuyaMCUOutputToChannel 5 1 5

setChannelType 6 toggle
linkTuyaMCUOutputToChannel 6 1 6

setChannelType 7 toggle
linkTuyaMCUOutputToChannel 102 1 7

setChannelType 8 toggle
linkTuyaMCUOutputToChannel 101 1 8
`