tuya-cloudcutter/tuya-cloudcutter #609: Error: NetworkManager is not running

tonysprenk, 2024-01-28:
I am getting the above error when running try-cloudcutter for the first time. I used the instructions for a headless raspberrypi0w. the pi is connected to my Mac using usb. any help would be greatly appreciated. Thanks!

Cossid, 2024-01-28:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md#pi-zero-2w-with-ssh-over-usb has instructions for installing Network Manger (which should be installed on Raspbian by default anyway), and after reboot, it should be active.  You can try running `sudo systemctl start NetworkManager` manually as well.

tonysprenk, 2024-01-28:
Thanks for the quick reply! I got it to work and its now installing the docker environment. its taking a long time and it seems stuck on Installing dependencies from Pipfile.lock is that normal?

Cossid, 2024-01-28:
Yes, that phase takes some time, especially on lower powered boards.  It takes about 5 minutes on a Pi4 for example.

tonysprenk, 2024-01-28:
ok good to know, thanks! It's been going for 20 minutes now, I'll grab another beer.

tonysprenk, 2024-01-28:
so 40 minutes now, is it still normal?

Cossid, 2024-01-28:
If it's still running, yes, we have seen reports of 0w's taking around 45-60 minutes.  Not much we can do about it.

tonysprenk, 2024-01-29:
So everything installed and I was able to run tuya-cloudcutter. Is it possible to install custom firmware on a device that uses a secondary MCU? Or is cloud cutting my only option? Also something else I ran into: after installing and testing everything I unplugged the pi to move the Mac closer to the tuya devices and after reboot it wouldn’t connect to the network through my Mac. In the system preferences under Ethernet gadget it says that the cable is unplugged even though it isn’t. How can I solve this? Or should I do the installation steps again?

Cossid, 2024-01-29:
Did NetworkManager stop again?  Try restarting it?  I don't have any experience with Pi Zeros and SSH over USB.

closed 2024-02-05 by Cossid

rdeserm, 2024-03-27:
I stumbled on this thread because I had same issue.  For posterity here is how I fixed it:

1) `sudo raspi-config`
2) Select `5 Advanced Options`
3) Select `A4 Network Config`
4) Select `NetworkManager`
5) `Reboot`

There is likely some console stuff you can type in to do this but this is what worked for me.

tonysprenk, 2024-03-27:
That’s what I did but after reboot the pi is not connected to the Mac anymore and ssh doesn’t work. It works if I do this without the reboot but cloudcutter cant find the tuya device for some reason.