tuya-cloudcutter/tuya-cloudcutter #611: New Device Request - Brilliant Frontier Twin Security Light

downriverbob, 2024-01-30:
I am trying to flash a Brilliant Lighting Frontier 20W Smart LED Twin Flood Light with Sensor (pictured below).

[image]

The MAC address lookup states it as a "Tuya Smart Inc" and firmware version is 1.0.5.
I tried all the 1.0.5 firmware options in cloudcutter but none have worked.
I tried to use Lightleak also but couldn't get it to work.

I have now disassembled and the WiFi chip is labelled as model WR3 (pictured below)

[image]

What other steps can I take to confirm supported or not before I go ahead and flash it overt serial?

Cossid, 2024-01-30:
WR3 is Realtek RTL8710B, and not supported by CloudCutter.  You would need to serial flash ESPHome/LibreTiny.

closed 2024-01-30 by Cossid

Cossid, 2024-01-30:
If you are able, please make a dump before flashing and post it here.  CouldCutter may possibly work with Realtek Tuya devices in the future, but there has been limited research into it so far.

downriverbob, 2024-02-04:
Hi Cossid,

Attached flash dump

[ltchiptool_ambz_2024-02-04_18-21-49.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14156057/ltchiptool_ambz_2024-02-04_18-21-49.zip)

Cossid, 2025-12-09:
Unfortunately, this firmware is [not currently vulnerable](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware#rtl8710bn) (common_115200 1.0.5)