tuya-cloudcutter/tuya-cloudcutter #612: New device request: Tuya PCIE Remote Switch Power/Reset

Technology89, 2024-01-31:
Hi, would be nice to add this device. Got a CBU in it. Dump is attached.

[dump_tuya_generic_pcie_power_trigger_31_01_2024.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14112503/dump_tuya_generic_pcie_power_trigger_31_01_2024.zip)
[image]
[image]

Cossid, 2024-01-31:
Added as `Tuya Generic PCIE Remote Switch`

closed 2024-01-31 by Cossid

BenJamesAndo, 2024-05-14:
Hey @Technology89 thanks for providing this dump. I flashed one of these devices today.

Would you happen to know if it's possible to control the status led and button on the device? For me sometimes the status led is on and other times it's off. Pressing the button doesn't do anything, unless I hold it for 5 seconds, then I get the following message `WIFI_RESET is not handled`. When pressing and holding the status led does flash.

This is my current config
```
esphome:
  name: pcie-remote-switch
  friendly_name: PCIe Remote Switch

bk72xx:
  board: cbu #BK7231N

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:

captive_portal:
    
web_server:

mdns:

uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600

tuya:
  # DPIDs processed from schema model: 000003cctr

switch:
  - platform: tuya
    switch_datapoint: 1
    id: pc_power
    name: Power

select:
  - platform: tuya
    enum_datapoint: 101
    name: Modereset
    optimistic: true
    options:
      0: Reset
      1: Forcereset
      2: "-"
```

This is the Tuya datapoints that's outputted
```
[16:52:53][C][tuya:041]: Tuya:
[16:52:53][C][tuya:056]:   Datapoint 1: switch (value: ON)
[16:52:53][C][tuya:062]:   Datapoint 101: enum (value: 2)
[16:52:53][C][tuya:074]:   Product: '{"p":"na90jncjlvw5t0ba","v":"1.0.0","m":0}'
```

Perhaps the status led and button aren't controllable from the cbu chip due to the MCU.