tuya-cloudcutter/tuya-cloudcutter #613: Process stopping while upgrading software

rafalolb, 2024-01-31:
I trying upgrade software to OpenBeken.

Log:
`rafal@raspberrypi:~/tuya/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -v
Building cloudcutter docker image
Sending build context to Docker daemon  32.86MB
Step 1/10 : FROM python:3.9.18-slim-bullseye AS base
 ---> 4034729d8940
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 35cfdf2342d6
Step 3/10 : FROM base AS python-deps
 ---> 35cfdf2342d6
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 3a23c875d0c8
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 3e3d5d67d44b
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> df5d1e65dd7e
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> ae569df0dc85
Step 8/10 : FROM python-deps AS cloudcutter
 ---> ae569df0dc85
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 6388bc7da738
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 821630a8772b
Successfully built 821630a8772b
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.80 - BK7231T / bk7231t_common_user_config_ty
   1.1.9 - BK7231N / oem_bk7231n_water_sensor_plus
   1.1.9 - BK7231T / bk7231s_common_iot_config_ty
   1.1.9 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.1.9 - BK7231T / oem_bk7231s_strip_ir_daybetter
 > 1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.2.1 - BK7231N / oem_bk7231n_dctrl_switch
   1.2.16 - BK7231N / oem_bk7231n_light_ty
   1.2.2 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.2.21 - BK7231N / oem_bk7231n_light_ty
   1.2.6 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.2.7 - BK7231N / oem_bk7231n_light_ty

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.262_bk7231n.ug.bin
[... 106 lines cut ...]
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 477, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 241, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmware=True)
  File "/src/cloudcutter/__main__.py", line 205, in __configure_local_device_or_update_firmware
    tornado.ioloop.IOLoop.current().start()
  File "/src/.venv/lib/python3.9/site-packages/tornado/platform/asyncio.py", line 205, in start
    self.asyncio_loop.run_forever()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 601, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1869, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/local/lib/python3.9/selectors.py", line 469, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Oh no, something went wrong with updating firmware! Try again I guess...
rafal@raspberrypi:~/tuya/tuya-cloudcutter $ 
`

I trying to upgrade 2 different devices.
Please help me,

Cossid, 2024-01-31:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

Cossid, 2024-01-31:
Read the FAQ, the app will not find the devices, but it will re-start the process of network connection, which CloudCutter should still be waiting for.

rafalolb, 2024-01-31:
I do'nt see this problem in FAQ, could you share directly link to this problem to FAQ?

Cossid, 2024-01-31:
It's the very one I already linked to, it says `The device will not connect to the Smart Life app, this is normal.`

rafalolb, 2024-01-31:
No no, It's not this step.

I'm on pount of instruction:
`Put the target device back into pairing mode. You can either continue using slow-blink mode, or use fast-blink mode here if it is more convenient.
`
I used fast-blinking mode, but I don't see device in my SmartLife app.
If I use slow-blinking mode my device broadcast a unsecured AP. I connected to it, but my SmartLife app don't see my device.

Cossid, 2024-01-31:
> I used fast-blinking mode, but I don't see device in my SmartLife app.

CloudCutter does not change the bluetooth functions, but I have seen other people state it just doesn't show up, in which case just use the slow blink method.

> If I use slow-blinking mode my device broadcast a unsecured AP. I connected to it, but my SmartLife app don't see my device.

That is normal, it won't see it, and the app will just keep sending data until it times out, but the device should instead be receiving instructions to connect to CloudCutter instead of SmartLife.  You're sending it the data for cloudcutter's AP afterall if following the instructions correctly.

rafalolb, 2024-01-31:
I did it :) I flashed 3 devices, but I don't know how :)

If process stopped in such line:
```
Jan 31 19:15:58 dnsmasq-dhcp[15]: 842012285 sent size: 4 option: 3 router 10.42.42.1
Jan 31 19:15:58 dnsmasq-dhcp[15]: 842012285 sent size: 4 option: 6 dns-server 10.42.42.1
```
I opened my SmartLife app and I clicked Add Device -> Add Manually -> Select my device (Breaker (Wi-Fi)) -> I Connect to 'cloudcutterflash' WiFi -> Clicked Next (I don't reset device) -> I hold the button for 5 second and I clicked Next -> Next -> Selected Blink Quickly -> I get message 'wrong network' -> selected my home network -> and I get info about countdown time. I waiting some times and on my logs from cloudcutter I get info about start flashing process.

Wow, thanks for tips.

closed 2024-01-31 by Cossid