tuya-cloudcutter/tuya-cloudcutter #614: New devices from LSC

septillion-git, 2024-02-04:
Hi,
I bought two LSC devices to play with:
LSC 3202086 XXL Led strip (CCT + RGB/C)
LSC 3202087 Power plug with energy monitoring

**3202086 Ledstrip**
The ledstrip has a bare BK7231N chip. In the Tuya app it shows as
Main module v1.0.18
MCU module v1.0.18

I was able to use cloudcutter on it by using "1.0.18 - BK7231N / oem_bk7231n_slide_strip_ty". Unfortunately there is no neopixel / RGB/C / WS2812 support in ESPhome to drive the RGB sections 😢 

Pinout:
Ctrl: P6 (main 24V+ to the strip, needed for pixels and CCT)
Cold: P9
Warm: P24
Button: P7
IR: P26 (untested)
Mic: P23 (untested)
Pixels: P16 (untested)
[Dumps.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14157452/Dumps.zip)

**3202087 Plug**
This uses a CB2S module with BK7231N and is the problem maker as it doesn't want to work with cloudcutter. It shows up in the Tuya app as:
Main module: v1.1.17
Other type: v1.0.4

For both modules I first attached a serial converter to them and extracted the .bin [Dumps.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14157452/Dumps.zip). But I'm not able to setup the toolchain for the profile building. Is anybody able to extract the profiles from here? And is it possible to add the plug to cloudcutter? Would be very cool if I don't need to open the plug again if I buy a second one. 😎

Cossid, 2024-02-04:
I have added a profile for the XXL LED strip as `LSC 3202086 XXL LED Strip v1.0.18`
Unfortunately for the plug, that one is patched, so we cannot profile a full CloudCutter profile for that.

Here is the pin data for the plug for reference for anyone serial flashing it:
```json
{
	"bt1_lv": 0,
	"bt1_pin": 7,
	"bt1_type": 0,
	"ch1_stat": 0,
	"ch_cddpid1": 9,
	"ch_dpid1": 1,
	"ch_num": 1,
	"chip_type": 0,
	"crc": 62,
	"ele_fun_en": 1,
	"ele_pin": 26,
	"ffc_select": 1,
	"jv": "1.0.4",
	"led1_lv": 1,
	"led1_pin": 6,
	"lose_vol": 180,
	"module": "CB2S",
	"net_trig": 2,
	"netled1_lv": 1,
	"netled1_pin": 10,
	"netled_reuse": 1,
	"over_cur": 17800,
	"over_vol": 264,
	"reset_t": 5,
	"resistor": 1,
	"rl1_lv": 1,
	"rl1_pin": 8,
	"rl1_type": 0,
	"sel_pin_lv": 1,
	"sel_pin_pin": 11,
	"vi_pin": 24,
	"vol_def": 0
}
```

closed 2024-02-04 by Cossid

Bart-1992, 2024-02-04:
@septillion-git thanks a lot for the pinout :) 

I also bought this 3202086 Ledstrip and installed "1.0.18 - BK7231N / oem_bk7231n_slide_strip_ty" and used cloudcutter succesfully.
Can you share your current ESPHOME config? 

thanks a lot

Cossid, 2024-02-04:
@Bart-1992 ESPHome doesn't really fully support it yet.  LibreTiny is still missing the SPI implementation for the bk72xx platform, and there is not currently a software driver that works for it.  Only the PWM white channels would currently work in ESPHome.  See https://upk.libretiny.eu/?profile=lsc-3202086-xxl-led-strip-v1.0.18 for that portion.  I believe OpenBeken supports the SPI portion somehow though.

septillion-git, 2024-02-04:
@Cossid Thank you very much! That was quick!

It's a pity the plug is patched 😒 That does make it a little bit less attractive to use. Although alternatives are slim as well. I'll try to get the thing running with ESPhome. Thank you very much for the pin data!

For the ledstrip, indeed no support for SPI / WS2812. I'm hopeful somebody will port it. But I was already glad I got a Beken chip working :) Having cloudless wifi devices got a lot harder after Tuya switched away from ESP's bit this give a little bit of it back. For now it goes into the unused smart home stuff pile 😁 

@Bart-1992 So yeah, if you want full effect of the strip you will need another chip. Not to hard to switch to an ESP but yeah, you will lose the nice controller thing. But as far as the ESPhome config:
```yaml
substitutions:
  device_name: lsc-xxl-ledstrip
  friendly_name: Study ledstrip

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

button:
  - platform: restart
    name: Restart

debug:
  update_interval: 30s

text_sensor:
  - platform: debug
    reset_reason:
      name: Reset Reason
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: uptime
    name: Uptime

output:
  - platform: libretiny_pwm
    id: output_cold
    pin: P9
  - platform: libretiny_pwm
    id: output_warm
    pin: P24

light:
  - platform: cwww
    id: light_cwww
    name: none
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    cold_white: output_cold
    warm_white: output_warm
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      - switch.turn_on: ctrl
    on_turn_off:
      - switch.turn_off: ctrl

binary_sensor:
  - platform: gpio
    name: "Button"
    pin: 
      number: P7
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_on: 50ms

switch:
  - platform: gpio
    pin: P6
    id: ctrl
    name: "ctrl"
    restore_mode: RESTORE_DEFAULT_OFF
```

Bart-1992, 2024-02-04:
@Cossid , 

Thanks. I've flashed the specified profile, but the P6 switch (power/ctrl) is missing? You must manually add it as GPIO switch to the esphome config to get some light out of it. 
Im will wait for the corresponding drivers to get implemented in LibreTiny. Or I might migrate to Openbeken if it takes really long. 

Again, many thanks for your support/help

JonathanTreffler, 2024-04-05:
For anyone that can make use of it, my variation of @septillion-git 's config, that uses the button as a power switch for the LED and does not send unnecessary entities to Home Assistant:

```
[...]

output:
  - platform: libretiny_pwm
    id: output_cold
    pin: P9
  - platform: libretiny_pwm
    id: output_warm
    pin: P24

light:
  - platform: cwww
    id: light_cwww
    name: Light
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    cold_white: output_cold
    warm_white: output_warm
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      - switch.turn_on: ctrl
    on_turn_off:
      - switch.turn_off: ctrl

binary_sensor:
  - platform: gpio
    name: "Button"
    internal: true
    pin: 
      number: P7
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_on: 50ms
    on_press:
    - then:
        - light.toggle: light_cwww

switch:
  - platform: gpio
    pin: P6
    id: ctrl
    internal: true
    name: "ctrl"
    restore_mode: RESTORE_DEFAULT_OFF
```