tuya-cloudcutter/tuya-cloudcutter #618: Ultrasonic Tuya water level sensor

hesham1216, 2024-02-09:
Hello Gentlemen,

I am not sure if I am asking the right questions (and / or) in the right place to be honest and excuse me if I am not :D.

But I want this device to be added in the device list, but I did not buy it yet. I wanted to confirm the potential of it being cloud cut and integrate with home assistant (sinse it has real time water level funcionality).

The device:
https://a.aliexpress.com/_oDAPG8A

Please let me know if it can be programmed so I may purchase it.

Thank you

Cossid, 2024-02-09:
Based on the fact it appears to have a 433mhz radio, there odds of it being Realtek are higher, and that would not be supported by CloudCutter.

closed 2024-02-09 by Cossid

hesham1216, 2024-02-10:
Then I guess I must DIY this water level sensor then 👍 
That ultrasonic device + lorea + esp32 may do the job