tuya-cloudcutter/tuya-cloudcutter #623: Firmware Dump - Tuya Smart Generic Touch Light Switch

cameroncros, 2024-02-18:
Bought: https://www.aliexpress.com/item/1005005732402318.html

Tuya Smart says its firmware v1.3.10, which is apparently unexploited. Not sure if firmware dumps are useful in that case, but I desoldered the IC and dumped it anyway.

[readResult_BK7231N_QIO_2024-18-2-14-38-38.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14321157/readResult_BK7231N_QIO_2024-18-2-14-38-38.zip)

Hopefully its useful, if not, apologies for the waste of time.

Cossid, 2024-02-18:
1.3.10 control switch firmware is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list.  Serial flashing will be your only option for this device.

closed 2024-02-18 by Cossid

cameroncros, 2024-02-18:
Yup, did dump and flash.

Is there value in receiving these known "safe" firmwares? If not, perhaps worth updating the readme?

Cossid, 2024-02-18:
They'd possibly be useful if there are any new future exploits, and since they require serial flashing, it's usually a good step to be taking anyway to potentially restore stock functionality if needed.