tuya-cloudcutter/tuya-cloudcutter #624: need help with generic smoke detector

digital-marine, 2024-02-18:
Hi there,

I'm too stupid to get the device flashed with another firmware or just use the standard firmware

```shell
markus@tc:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -r -v
Building cloudcutter docker image
[+] Building 1.5s (13/13) FINISHED                                                                docker:default
 => [internal] load build definition from Dockerfile                                                        0.0s
 => => transferring dockerfile: 462B                                                                        0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                              1.2s
 => [internal] load .dockerignore                                                                           0.0s
 => => transferring context: 2B                                                                             0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:c59d6aaacd1e721ffebfa50ab8991625c  0.0s
 => [internal] load build context                                                                           0.1s
 => => transferring context: 3.02kB                                                                         0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rf  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                               0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                         0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                    0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                  0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                   0.0s
 => exporting to image                                                                                      0.0s
 => => exporting layers                                                                                     0.0s
 => => writing image sha256:85beaba09cebacf3403504246e9dea9a9a80fe3b80162563e392b34667f5a551                0.0s
 => => naming to docker.io/library/cloudcutter                                                              0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.0.5 - BK7231N / dy_yg400a
   1.0.4 - BK7231N / tiandu_bk7231n_light2_strip
   1.0.4 - BK7231T / cuco_sl1_b_bk
   1.0.4 - BK7231T / oem_bk7231s_light4_4to5_deppon60A
   1.0.4 - BK7231T / oem_bk7231s_light_strip_ew010_jbtled
   1.0.4 - BK7231T / oem_bk7231s_ty_ffc_newdp
   1.0.5 - BK7231N / bk7231n_lowpower_common_ty
 > 1.0.5 - BK7231N / dy_yg400a
   1.0.5 - BK7231N / oem_bk7231n_dctrl_switch
   1.0.5 - BK7231N / oem_bk7231n_light_atomi
   1.0.5 - BK7231T / oem_bk7231s_strip_ty
   1.0.6 - BK7231N / oem_bk7231n_pir_tuya
   1.0.6 - BK7231N / oem_bk7231n_strip_rgb_diwen
   1.0.6 - BK7231T / oem_bk7231s_dltj_2switch_alerc

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 > ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231n.ug.bin

Selected Device Slug: tuya-generic-tysm-01-smoke-detector
Selected Profile: dy-yg400a-1.0.5-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
[... 435 lines cut ...]
  File "/src/cloudcutter/__main__.py", line 477, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 241, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmware=True)
  File "/src/cloudcutter/__main__.py", line 205, in __configure_local_device_or_update_firmware
    tornado.ioloop.IOLoop.current().start()
  File "/src/.venv/lib/python3.9/site-packages/tornado/platform/asyncio.py", line 205, in start
    self.asyncio_loop.run_forever()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 601, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1869, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/local/lib/python3.9/selectors.py", line 469, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Oh no, something went wrong with updating firmware! Try again I guess...

``` 
Without using Smart Life / Tuya App the script hang at "MQTT Connected". As the access point changed to A- it seems the cut worked, but that's it. I hit CTRL-C after about an hour.

KIW-L21 is the hostname of my Android Phone running LineageOS 16 (Android 9), which is my only device with 2.4GHz network only.

I run the script as normal user and root, slow blink mode, fast blink mode and any other options I've found.

Any help would be appreciated.

Cossid, 2024-02-18:
It's not just you.  I have this same device, and I can't seem to get it to finish the exploit OTA either.  Battery powered devices with power saving features seem to react differently to network join commands.  Sometimes you can keep them alive by pressing a wake button repeatedly, but because this device has a buzzer, it gets annoying trying the different ways pretty quickly.

I'd probably just recommend serial until it becomes clearer what keeps this specific device awake.

Also note that, deep sleep is not well supported on ESPHome yet, so this device will only last about a day on batteries with current support (there are deep sleep patches being worked on, but not complete yet).

digital-marine, 2024-02-18:
It's a shame that battery-powered devices aren't well supported yet, but it's nice that I'm not too stupid .

closed 2025-01-09 by Cossid