tuya-cloudcutter/tuya-cloudcutter #627: Firmware dump : Action LSC Smart plug with power monitoring firmware dump 1.3.5 tuya

FlorianSoler, 2024-02-24:
### Hello, 
I have bought the latest version of the action smart plug :
https://www.action.com/nl-nl/p/3202087/lsc-smart-connect-slimme-stekker/

Tuya Smart indicate its firmware v1.3.5.
[readResult_BK7231T_QIO_lsc_1.3.5_tuya-smartplug_2024-21-2-22-19-41.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14393203/readResult_BK7231T_QIO_lsc_1.3.5_tuya-smartplug_2024-21-2-22-19-41.zip)

board revision : WP02GE 2023-09-05
tuya board : WB2S
mcu : BK7231T

Cossid, 2024-02-24:
Added as `LSC 3202087 Plug with Energy Monitoring v1.3.5`

closed 2024-02-24 by Cossid

Xenuzz, 2024-05-21:
Hello, I have the same socket but the firmware is in the main module: V1.1.17 and in the MCU module: V1.0.4. My problem is that I can't flash this socket and I don't have any updates for this socket either. What can I do now? Thanks in advance.

Cossid, 2024-05-21:
1.1.17 is on the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware), serial flashing will be the only option for that device.