tuya-cloudcutter/tuya-cloudcutter #629: Lexi String Light Controller

BenJamesAndo, 2024-02-29:
https://www.lexilighting.com.au/products/wifi-tuya-app-control-adaptor

This device is used to control 2 wire string lights common on Christmas trees or houses. It has a WB2S chip.
[image]

Despite opening it up with a rotary tool, I ended up using Lightleak to get the dump file. It's still running Tuya as I'm not sure how to flash it without a working profile. Lightleak did generate a profile but it doesn't seem to work. None of the existing profiles on cloud-cutter worked either, hence me using Lightleak.
Main Module 1.1.0
MCU Module Version: 1.1.0

[dump_2024-02-29_09-43-48_lightleak.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14441182/dump_2024-02-29_09-43-48_lightleak.zip)
[profile.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14441189/profile.json)

Cossid, 2024-02-29:
Added as `Lexi APT01 String Light Controller`

closed 2024-02-29 by Cossid

Cossid, 2024-02-29:
Note: the device configuration came from the firmware and not the storage, and may just be the demo light configuration.  If it is 2-wire, I'm not sure what options you currently have.  OpenBeken may support it.  LibreTiny doesn't currently support SPI or I2C.  You might be best trying cut LocalTuya first, unless you're pretty comfortable with handling less-supported devices.

BenJamesAndo, 2024-02-29:
Thank you!
Yes the ESPHome YAML that it output was wrong. There's no RGB functionality. It did somewhat work out of the box with OpenBeken. This device uses a H-Bridge to control the lights. Using ESPHome YAML from [this post](https://community.home-assistant.io/t/h-bridge-2-wire-fairy-light-controller/633484) I was able to get it working quite nicely.

Everything works except for the microphone as it seems to not be supported yet. Below is my code. Is it worth me putting it somewhere more official?
Physical button is P8 and status LED is P7.

```
esphome:
  name: lexi-controller
  on_boot:
      priority: 600
      then:
        - light.turn_on:
            id: ledstrings
            brightness: 100%
            color_temperature: 4268K

bk72xx:
  board: generic-bk7231t-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  password: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

output:
  - platform: libretiny_pwm
    id: pina
    pin: P24
  - platform: libretiny_pwm
    id: pinb
    pin: P26

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P8
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - light.toggle: ledstrings

light:
  - platform: status_led
    id: stat_led
    name: "Switch state"
    pin:
      number: P7
  - platform: hbridge
    id: ledstrings
    name: "led-strings"
    pin_a: pina
    pin_b: pinb
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
    - light.turn_on: stat_led
    on_turn_off:
    - light.turn_off: stat_led
    effects:
      - strobe:
          name: "Flash A-B 500ms"
          colors:
            - state: true
              color_temperature: 2000 K
              duration: 500ms
            - state: true
              color_temperature: 6534 K
              duration: 500ms
      - strobe:
          name: "Flash A-AB-B 500ms"
          colors:
            - state: true
              color_temperature: 2000 K
              duration: 500ms
            - state: true
              color_temperature: 3500 K
              duration: 500ms
            - state: true
              color_temperature: 6534 K
              duration: 500ms
            - state: true
              color_temperature: 3500 K
              duration: 500ms
      - lambda:
          name: "Fade A-AB-B 10s"
          update_interval: 10s
          lambda: |-
            static int state = 0;
            auto call = id(ledstrings).make_call();
            call.set_transition_length(10000);
            if (state == 0) {
              call.set_warm_white(1.0);
              call.set_cold_white(0.0);
            } else if (state ==1) {
              call.set_warm_white(0.0);
              call.set_cold_white(1.0);
            }
            call.perform();
            state += 1;
            if (state == 2)
              state = 0;
      - lambda:
          name: "Slow-Glo | Fade A-AB-B 2s 0.2"
          update_interval: 2s
          lambda: |-
            static int state = 0;
            auto call = id(ledstrings).make_call();
            call.set_transition_length(2000);
            if (state == 0) {
              call.set_warm_white(1.0);
              call.set_cold_white(0.2);
            } else if (state ==1) {
              call.set_warm_white(0.2);
              call.set_cold_white(1.0);
            }
            call.perform();
            state ++;
            if (state == 2)
              state = 0;
```

Cossid, 2024-02-29:
> Is it worth me putting it somewhere more official?

Isn't really an official place for something like this, so here is as good as any.