tuya-cloudcutter/tuya-cloudcutter #631: Brilliant Smart WiFi 6 power outlets with USB

BenJamesAndo, 2024-03-01:
https://brilliantlighting.com.au/product/wifi-cannes-power-board-04c232
Chipset BK7231N
Tuya 1.1.8

Each power outlet can be controlled. The USB ports are grouped as the 7th outlet.
[BK7231N_QIO_obk08CADA4C.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14457491/BK7231N_QIO_obk08CADA4C.zip)

Cossid, 2024-03-01:
This dump appears to be of OpenBeken, not the original Tuya firmware.

closed 2024-03-01 by Cossid

BenJamesAndo, 2024-03-01:
Ahh appears I made the same mistake as with https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/597
I did the 2MB dump from OpenBeken web UI after flashing it. I read that Lightleak mainly only works with BK7231T and not BK7231N chips, is that still the case? If so then would the only way to have gotten a decent dump would be to open up the board and do it via serial cables?

Cossid, 2024-03-01:
> I read that Lightleak mainly only works with BK7231T and not BK7231N chips, is that still the case?

Not really, it's spotty with both, depending on the SDK of the device (something you can't know in advance), but T has 1 LightLeak profile, and N has 3, so you may just have to try multiple times (and I'd try twice with each profile, as that has had better luck for me).