tuya-cloudcutter/tuya-cloudcutter #632: Antela Wifi smart plug with PM formware extract

arly69, 2024-03-05:
This is extracted storage area json from itchiptool for
Antela model SPL-W-TY-PM-FR-RY 
CB2S for control, ZX Wing WP02GE platform with BL0937 for Power Measurement
manufactured by ZX wing : https://www.nb-royoung.com/german-type-wi-fi-smart-plugwp02ge-product/

[Antela_WifiSmartPlugWithPM_Model_SPL-W-TY-PM-FR-RY_storagedata.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14498972/Antela_WifiSmartPlugWithPM_Model_SPL-W-TY-PM-FR-RY_storagedata.json)

Cossid, 2024-03-05:
Are you able to provide the full dump?

arly69, 2024-03-06:
Yes
[ANTELA firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14503501/ANTELA.firmware.zip)

Cossid, 2024-03-06:
Added as `Antela SPL-W-TY-PM-FR-RY Plug with Energy Monitoring CB2S v1.1.8`

closed 2024-03-06 by Cossid

arly69, 2024-03-06:
yes
[ANTELA firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14503610/ANTELA.firmware.zip)