tuya-cloudcutter/tuya-cloudcutter #635: New BK7231N with MainModule v1.3.10

rafalolb, 2024-03-11:
Hello,

I used your tuya-cutter more times works great.

Today I get from China new Beker 7231N (1 gang switch). I Installed it in TuyaApp and I checked MainModule version and I see v1.3.10.

Can you add support to this version?
Can I use other version (for example 1.3.1) on tuya-cutter?

[image]
[image]
[image]

Cossid, 2024-03-11:
Unfortunately, 1.3.10 is on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list

Serial flashing will be your only option for that device.

closed 2024-03-11 by Cossid

blka, 2025-12-15:
hey @rafalolb did you find correct UART connection pins for this board maybe?