tuya-cloudcutter/tuya-cloudcutter #638: Patch Tuya wifi switch without beken module

rafalolb, 2024-03-21:
Hey,

can I patch WIFI Switch Module (model: QS-WIFI-SO5) by Tuya-CloudCutter?

My Wifi switch module has insite ESP module (not Beken module).

Specs:
https://templates.blakadder.com/QS-WIFI-S05.html

[image]

Cossid, 2024-03-21:
No, ESP is not currently support.  While technically vulnerable to the same exploit, a working exploit path is not currently know (and may or may not even be possible)

closed 2024-03-21 by Cossid

rafalolb, 2024-03-21:
Can I check?

Cossid, 2024-03-21:
No, CloudCutter doesn't support a path for ESP, there are no profiles that would work with it, and no support for flashing that type of firmware.  Unless you know assembly well, there isn't anything you can do.

You can try Tuya-Convert or serial flashing.

rafalolb, 2024-03-21:
Tuya-convert is old repository and not works in my device :( I'm afraid to solder, I'm developer 🙂

I want to try flashing by your cutter.

Cossid, 2024-03-21:
Again, unless you know assembly and building stack redirection, there is no other starting point.  As ESP is Xtensa and not ARM Thumb, we're not even sure if the stack redirection is even possible, we don't have anyone knowledgeable on that subject.

As-Is, CloudCutter is not designed to work with ESP.