tuya-cloudcutter/tuya-cloudcutter #639: Error: Connection activation failed: 802.1X supplicant took too long to authenticate.

rolandbreedveld, 2024-03-23:
I try to flash a LSC with a Raspberry Pi (both a Pi3 and a Pi4), but it results in this error, any idea how to fix this ?
LSC smart plug 2578685 device version 1.1.8

./tuya-cloudcutter.sh -s accessxxxxxxx XXXXXXX -v -r  -p lsc-2578685-970766-smart-plug-cb2s-v1.1.8

Selected Device Slug: lsc-2578685-970766-smart-plug-cb2s-v1.1.8
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP

Found access point name: "SmartLife-1E36", trying to connect...
Device 'wlan0' successfully activated with 'c52a0541-20e4-45ad-897f-995d2xxxx1ee'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/TMhC6BNzjx4w.deviceconfig
Saved device config in /work/configured-devices/TMhC6BNzjx4w.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-1E36", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
Wiping NetworkManager configs
.........................
Found access point name: "SmartLife-1E36", trying to connect...
Device 'wlan0' successfully activated with '885e9e32-7a01-469f-a6dd-84d8xxxxbd66'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

Cossid, 2024-03-23:
`Connection activation failed: 802.1X supplicant took too long to authenticate.` is just a general network timeout, and sometimes that's because the system has a scan from before the device was reset, and tries to connect too early.  But it always follows it up, and in this case connects to an unexploited device.

Are you certain your device is Beken, and certain it is version 1.1.8?  LSC did transition from ESP to Beken like most
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out
It's also possible it could be a Realtek device.

Either way, the selected profile did not appear to work with your device.

rolandbreedveld, 2024-04-03:
Tried it on a intel/ubuntu host, with both devices same issue:
./tuya-cloudcutter.sh -s access0_xxx xxxxxx -v -r -p lsc-2578685-970766-smart-plug-cb2s-v1.1.8
./tuya-cloudcutter.sh -s access0_xxx xxxxxx -v -r -p lsc-3001686-970709-warm-white-smart-filament-e27-806lumen
checked, versions are exactly correct, both checked on the box and in the Tuya app.

On tis pc, no 802.1X error:

Selected Device Slug: lsc-3001686-970709-warm-white-smart-filament-e27-806lumen
Selected Profile: oem-bk7231s-light-ty-2.9.6-sdk-1.0.2-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP

Found access point name: "SmartLife-9872", trying to connect...
Device 'wlan1' successfully activated with '28411b7f-2237-4986-b3a1-9f535f111864'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/zXEO6HHyyj27.deviceconfig
Saved device config in /work/configured-devices/zXEO6HHyyj27.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
....
Found access point name: "SmartLife-9872", trying to connect...
Device 'wlan1' successfully activated with '28411b7f-2237-4986-b3a1-9f535f111864'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

rolandbreedveld, 2024-04-03:
ok, thanxzzz
both the plug and the bulb have a mac address from [Tuya Smart Inc]

and this is the log for the plug, i double checked, the serial number and software version:

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 20360.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

Selected Device Slug: lsc-2578685-970766-smart-plug-cb2s-v1.1.8
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
................................................................................
Found access point name: "SmartLife-1E36", trying to connect...
Device 'wlan1' successfully activated with 'fd9e6b85-9812-4fa4-beb2-5a3bc8a8101e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/12gpEb9WW3h2.deviceconfig
Saved device config in /work/configured-devices/12gpEb9WW3h2.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
......................................................................................
Found access point name: "SmartLife-1E36", trying to connect...
Device 'wlan1' successfully activated with '1490773c-24d7-4aaa-a1b9-b8ce0956fc47'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

rolandbreedveld, 2024-04-03:
after the last row, both the bulb and the plug, stay in AP mode (slow-blinking)
both are pingable at 192.168.175.1

Cossid, 2024-04-03:
These devices likely have different modules than the existing profiles.

You can try by firmware version / name.  For the plug, try 1.1.8 oem_bk7231s_rnd_switch instead.
You haven't mentioned which firmware version your bulb actually has, so I can't make a recommendation there.

How have you verified what version your plug is?  I'm specifically talking about the firmware version, not the model number.

closed 2024-05-05 by Cossid