tuya-cloudcutter/tuya-cloudcutter #641: [DUMP]alfawise pe1004t

arnoakavdb, 2024-03-26:
Hi,
i dump the original firmware of the smart plug socket alfawise pe1004t because i find nothing about firmware 0.1.3 on tuya cloud cutter.if someone can do something with it...
[image]

[Alfawise_PE1004T_0.1.3.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14758977/Alfawise_PE1004T_0.1.3.bin.zip)

Cossid, 2024-03-26:
This dump is for an ESP8266 device, which is not supported by CloudCutter.  You can try Tuya-Convert or serial flashing for this device.

closed 2024-03-26 by Cossid

arnoakavdb, 2024-03-26:
thanks i already use serial flashing for 1 i'll try tuya convert for the 2 others

Cossid, 2024-03-26:
The compile date `Jul 27 2018` in the firmware you posted suggests it should be vulnerable to Tuya-Convert, as that wasn't patched until late 2019.