tuya-cloudcutter/tuya-cloudcutter #643: KMC-30154

powersupplyf, 2024-04-02:
https://www.elektroda.com/rtvforum/viewtopic.php?p=21023310#21023310

Following
1) Note about adding new profiles #203
https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/203
brings me to

2) https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io

3) clicking on "Pull requests"

4) brings me to "There aren’t any open pull requests."

5) but offers me a "New pull request"
clicking that

6) brings me to "Compare changes"
What/which changes do I compare to???
I'm trying to add a NEW unit (KMC-30154), I checked and it doesn't exist yet, so what do I compare??

7) To be able to continue however, it then needs me to select
"Choose different branches or forks above"

8) gh-pages or master@{1day}...master
So which do I choose there???

9) "About pull requests" didn't help at all

This is a very frustrating thing and I wonder if it's worth it to go through the whole ordeal to even make a dump and then try to "add" anything?

Cossid, 2024-04-02:
To do it that way, you create a fork of the `tuya-cloudcutter.github.io` repo, create a branch, add the files to your branch, and then start the pull request from there.

But you can just post the full dump here as well, and we will generate the profile.

powersupplyf, 2024-04-03:
I understood the last sentence.
here it is:

[Smart.Plug.KMC.30154.readResult_BK7231T_QIO_sdsdf_2024-28-3-07-19-17.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/14845294/Smart.Plug.KMC.30154.readResult_BK7231T_QIO_sdsdf_2024-28-3-07-19-17.zip)

I noticed that the LED if set to WiFI only blinks when no connection.
When set to "LED_n" it turns on/off with the relay
The original would however use the LED for both functions.
Is it possible to set that here, too?

Thank you for your great tool and help with all this!!

Cossid, 2024-04-03:
Added as `KMC 30154 Smart Plug Slim v1.0.6`

> I noticed that the LED if set to WiFI only blinks when no connection.
> When set to "LED_n" it turns on/off with the relay
> The original would however use the LED for both functions.
> Is it possible to set that here, too?

That depends on the firmware you install.  _n is just inverting the state.  With ESPHome you can add a `light:` entity with a platform of `status_led` instead of just a simple `status_led:` entity and it will do double duty, where it will blink for the wifi pattern and be controlled as a light manually with events.  I can't speak as to how to do that with OpenBeken, which I'm assuming you're using based on the LED_n statement.

closed 2024-04-03 by Cossid

powersupplyf, 2024-04-03:
Hi Cossid, thank you for your quick reply.
Yes, using OpenBeken unfortunately, Tasmota seems nearly non existent anymore.
Will check over there what to do.
https://www.elektroda.com/rtvforum/viewtopic.php?p=21030164#21030164
Do you think if there's a way to link/centralize the json profiles from OpenBeken and tuya-cloudcutter? (and maybe others??).
It seems such a waste of resources and time...
Thanks again for your great tool, I just used it (took me some hours to get it right) to OTA one unit.

Cossid, 2024-04-03:
OpenBeken uses the CloudCutter feed here: https://openbekeniot.github.io/webapp/devicesList.html - you must choose Cloudcutter Profile then filter to your device, and hit the config button.  I can't speak to it further than that, as I don't use OpenBeken.
ESPHome also uses the CloudCutter feed here: https://upk.libretiny.eu/?profile=kmc-30154-smart-plug-slim-v1.0.6

Some devices don't have configuration data included though, and each platform must fill those gaps.  This device does have basic configuration data though.

powersupplyf, 2024-04-03:
Indeed there are some units I dumped and was missing from beken, Now I understand, never knew that selection, thanks!

agibson-fl, 2024-05-05:
@Cossid I put an update on the thread: 
https://www.elektroda.com/rtvforum/viewtopic.php?p=21070350#21070350
My theory there was correct about the 30154 having two different firmwares. 

Is there a way to make any further updates to the 30154 profile to include both 1.0.6 and 1.0.7 so you can just go to the KMC menu to begin with? Not sure if that's worth it, but the reality is there are two firmware versions.

Here were the steps I took to get both flashed using cloudcutter:

Some devices must have been on 1.0.6 firmware:
---------------------------------------------
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 ► By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: KMC
   Horevo
   Hykolity
   IWIIK
   Immax
   Jiangsu Barep
   KHSUIN
 ► KMC
   KOGAN
   Kanlux
   Klarstein
   Kobi
   Kuled
   LEDLUMEN

[?] Select the article number of your device: 30154 Smart Plug Slim v1.0.6
   30153 Smart Plug Mini
 ► 30154 Smart Plug Slim v1.0.6
   30407 Smart Tap Mini 4 Outlet Plug BK7231N v1.1.8
   30407 Smart Tap Mini 4 Outlet Plug BK7231T v1.1.8

[?] Select the firmware version and name: 1.0.6 - BK7231T / oem_bk7231s_rnd_switch
 ► 1.0.6 - BK7231T / oem_bk7231s_rnd_switch

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
 ► ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
   OpenBeken-v1.17.262_bk7231t.ug.bin

Selected Device Slug: kmc-30154-smart-plug-slim-v1.0.6
Selected Profile: oem-bk7231s-rnd-switch-1.0.6-sdk-2.0.0-30.05
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

Some devices were on 1.0.7 firmware:
---------------------------------------------

Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
By manufacturer/device name
► By firmware version and name
From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.0.7 - BK7231T / oem_bk7231s_rnd_switch
1.0.6 - BK7231T / oem_bk7231s_dltj_2switch_alerc
1.0.6 - BK7231T / oem_bk7231s_fan_control_arlec
1.0.6 - BK7231T / oem_bk7231s_light_db_par38
1.0.6 - BK7231T / oem_bk7231s_rnd_switch
1.0.6 - BK7231T / realme_BK7231T_plug_rmh2009_un
1.0.7 - BK7231N / fresco
► 1.0.7 - BK7231T / oem_bk7231s_rnd_switch
1.0.7 - BK7231T / oem_bk7231s_rnd_zero_dctrl_switch
1.0.8 - BK7231N / oem_bk7231n_light3_string_jt
1.0.8 - BK7231T / oem_bk7231s_light2_pir_mk
1.0.8 - BK7231T / oem_bk7231s_light_db_par38
1.0.8 - BK7231T / oem_bk7231s_light_ty
1.0.8 - BK7231T / oem_bk7231s_rnd_switch

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231T chip: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
► ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
OpenBeken-v1.17.262_bk7231t.ug.bin

Selected Device Slug: kmc-30153-smart-plug-mini
Selected Profile: oem-bk7231s-rnd-switch-1.0.7-sdk-2.0.0-30.06
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin

Cossid, 2024-05-05:
Added a v1.0.7 profile for this device.