tuya-cloudcutter/tuya-cloudcutter #646: FYI Only - Long build times on RPi 1B

Atreidae, 2024-04-07:
Hi team. I had a brainfart whilst I was cutting a few of my devices and pulled an old RPi.
I successfully got Cloudcutter running on an RPi 1B with a USB wifi adapter. I figured I'd just drop you a line to let you know the build time was FOUR HOURS.

[image]

So yeah for anyone looking for "Installing dependencies from Pipfile.lock"
Yes, it can take a freakishly long time to build the docker image. Especially if you are silly like me and use ancient hardware.
But things are a lot faster on re-runs once the image is built

The only clue I had was this issue mentioning 15 minute build times on Rpi 3's
https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/559

Cossid, 2024-04-07:
Make sure you have the latest checkout, as there was some cleanup done around the time of the other issue you linked that had improvements for build times, but yeah, RPi 1, like the Zero line is going to be slow.

closed 2024-04-07 by Cossid

Atreidae, 2024-04-08:
Yeah it was a recent clone. The ticket was more for anyone that decided to be silly like me and went searching for answers like I did.. I saw 15 minutes and figured "yeah okay. but its been 2 hours" 

It's amazing to see how much faster modern Rpi's are.

I will say the project has given me a lot of confidence in getting a large batch of Tuya in wall switches and playing with them!

Cossid, 2024-04-08:
Just be aware that many recently manufactured products are patched these days.  You can always serial flash, provided module access to the correct pins (some are not easily possible though).