tuya-cloudcutter/tuya-cloudcutter #647: LSPS5 2-gang profile

ljovcheg, 2024-04-09:
Hi,

I have Tuya 2-gang switch. I've tried all of 2-gang profiles and none worked form me.
According to info, the model is LSPS5. https://www.aliexpress.com/item/1005003379628923.html

Any suggestions? 

Thanks

[image]

ljovcheg, 2024-04-09:
Forgot to mention, I have the same model but with 1 gang and "DS101-JL 1 Gang Smart Switch" profile worked perfectly for me

Cossid, 2024-04-09:
The same profile for the 1-gang should work for the 2-4 gang as well as long as the version matches.

ljovcheg, 2024-04-09:
I am such an idiot. Thank you for opening my eyes.

closed 2024-04-09 by ljovcheg