tuya-cloudcutter/tuya-cloudcutter #650: New Device: Moes 3gang Switch

KyleStilkey, 2024-04-18:
I'm going to take a long shot that this device is patched and it will need to be serial flashed but just in case, I will still provide the dumped rom for the device anyways.

[MOEs Switch.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15019178/MOEs.Switch.zip)

https://www.amazon.com/gp/product/B0C3HD3LQH/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Cossid, 2024-04-18:
This is switch firmware v1.3.5, which is on the known patched list.  Serial flashing will be the only option for this device.

The following is the device configuration if anyone needs it for generating a profile.
```json
{
	"backlit_dp": 16,
	"backlit_select": 0,
	"bt1_lv": 0,
	"bt1_pin": 23,
	"bt2_lv": 0,
	"bt2_pin": 9,
	"bt3_lv": 0,
	"bt3_pin": 28,
	"bt_type": 0,
	"cd_flag2": 1,
	"ch_cddpid1": 7,
	"ch_cddpid2": 8,
	"ch_cddpid3": 9,
	"ch_dpid1": 1,
	"ch_dpid2": 2,
	"ch_dpid3": 3,
	"ch_flag1": 1,
	"ch_flag3": 1,
	"ch_num": 3,
	"crc": 118,
	"cyc_dpid": 17,
	"inch_dp": 19,
	"init_conf": 14,
	"jv": "1.1.0",
	"led1_lv": 0,
	"led1_pin": 14,
	"led2_lv": 0,
	"led2_pin": 16,
	"led3_lv": 0,
	"led3_pin": 20,
	"led_dp": 15,
	"light_status_select": 0,
	"module": "CBU",
	"net_trig": 4,
	"rand_dpid": 18,
	"reset_t": 5,
	"rl1_lv": 1,
	"rl1_pin": 24,
	"rl2_lv": 1,
	"rl2_pin": 26,
	"rl3_lv": 1,
	"rl3_pin": 6,
	"total_stat": 0,
	"zero_select": 0
}
```

closed 2024-04-18 by Cossid

KyleStilkey, 2024-04-18:
Had a feeling it was a patch build probably why someone returned these to Amazon lol but easy enough for me to flash. I have also added your details to the Discord. Thanks again!

starbrights, 2024-04-23:
Already closed but may I ask:
I do have a Moes Curtain Switch (Software 1.1.0).
Tried different devices with 1.1.0 for BK7231N, but always get:
[!] The profile you selected did not result in a successful exploit.

Can you help or should I better open a new task? Howto create a dump?

starbrights, 2024-04-23:
Well, I see that there is an imprint on housing CB-CS03 and CB3S is printed at the board (near antenna). That imho indicates an BK7231N. 
Shouldn't this be the proper profile than:  1.1.0 - BK7231N / oem_bk7231n_curtain_switch ?
That is what I tested several time.  Or do I have other options?

Regarding dump: I don't have a serial connection to the board, I am not good in soldering ;) So only lightleak would be an option, right?

Cossid, 2024-04-23:
Correct, CB3S is BK7231N, and if that profile didn't work then a dump will be needed.  You are free to try LightLeak, but there is no guarantee it will work.

starbrights, 2024-04-23:
Do it understand it right: to get (with some luck) a dump I need:
* Android phone with cloudcutter-android app, 
* my target device (here Moes) and 
* and additional mini-board with  platformio-custom-ap on it?

Cossid, 2024-04-23:
That is correct, yes.  You'll need to manually grant the android app location/wifi access I believe.

Cossid, 2024-04-23:
This isn't an arduino sketch, it is platformio, which you must use directly.  If you're unfamiliar, the easies way is Visual Studio Code with the PlatformIO plugin.

starbrights, 2024-04-24:
Found an easier way: There are prebuilds on that page. 
https://github.com/tuya-cloudcutter/lightleak/releases/tag/v2024.2.9
Without really nowing what I do:
esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB -ff 80m -fm dio 0x00000 lightleak-custom-ap_v2024.2.9_esp8266.bin
At least I get an AP LightleakIdle. 
I enabled permission for the app. But handling is not clear to me.
It seems I need first to have ethernet access to load profiles and devices.

First of all: Am I right - my smartdevie has to be in slow-blinking mode?

But than it ask for device state? It recommends first option.
Its is starting but always fail at "Establish connection with device" (device is blinking slowly).

I am not sure what this Classic profiles are for? Selecting one of it returns to profile selection again, until I chose one of the Lightleak profiles (and there are 12 for BN8321N).

I could once come further in the list of stages, but this is by chosing option 2 in stated (configured to join customAP).  It goes until Check if device is exploitable - but that was without the customAP (the esp8266 board with lightleak Wifi) ?!?!

I don't understand the procedure. 
Chosing Device State: unconfigured: Android is connecting to customAP (lightleak)
Chosing Device State; Confiigured to join CustomAP: Android joins the SmartDevice network (SmartLife-xxx)

Cossid, 2024-04-24:
Yes, the device must be in slow blink mode, even for LightLeak.
For lightleak, just ignore any profile that doesn't explicitly say LightLeak in it.  It shares profiles with CloudCutter, but can't use CloudCutter's profiles.
You can't skip states, as it leads to incorrect assumptions.
The middle device should change from LightLeakIdle to LightLeakCustom at some point early in the process.

starbrights, 2024-04-24:
Mh, but using state 1 (unconfigured) is right? The others states can't use for dump?
Did I understood you right: the ESP (CustomAP) changes its WiFi net from LightLeakIdle to LightLeakCustom? Need to check that. Maybe I need to try another phone, my one is rather old Android 8.
And I need to check all the 12 of BK7321N profiles, right?

Cossid, 2024-04-24:
Yes, you should pretty much always start from unconfigured with each new profile you try.
The first thing that happens is the phone should connect to LightLeakIdle and a successful connection will change that to LightLeakCustom and start trying to connect to the device.
Yes, you may potentially need to try all 12 N profiles, assuming you are sure this is an N device (you can't tell that by the firmware version number).

starbrights, 2024-04-25:
Hi, with my tablet it works just fine. 
[dump_2024-04-25_11-34-04_lightleak.tar.gz](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15107665/dump_2024-04-25_11-34-04_lightleak.tar.gz)
I hope that can be used to create a profile for cloudcutter.

Cossid, 2024-04-25:
Please create a new issue and attach it there, so it can be properly linked, include details of what the device is.