tuya-cloudcutter/tuya-cloudcutter #652: Flash dump of Yieryi YY-W9099 water meter

LiesOfOmission, 2024-04-20:
Hi, here's a flash dump of some generic Chinese water meter that measures ph/ec/temperature. The chip is a CB3S with BK7231N: [water-meter-firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15049385/water-meter-firmware.zip)

Looking around, I see a bunch of seemingly identical devices under the name of Kmoon, Irfora, Yieryi, and probably more - they all seem to be exactly the same: https://shop.yieryimeters.com/products/yieryi-smart-bluetooth-water-quality-tester-6-in-1-ph-tds-ec-salinity-s-g-temp-meter-wifi-app-intelligent-control

It's using TuyaMCU to get ph/ec/temp data from some other chip on the board, BUT I have no idea yet which datapoints map to which values :confused: Is that something this project could give me some guidance on?

I guess the ideal outcome is a new cloudcutter profile for this device so others can flash without cracking it open :crossed_fingers:

LiesOfOmission, 2024-04-20:
I've already flashed it with ESPHome, here's the datapoints it sees:

```
[20:49:25][C][tuya:041]: Tuya:
[20:49:25][C][tuya:058]:   Datapoint 118: int value (value: 910)
[20:49:25][C][tuya:058]:   Datapoint 119: int value (value: 0)
[20:49:25][C][tuya:058]:   Datapoint 120: int value (value: 300)
[20:49:25][C][tuya:058]:   Datapoint 121: int value (value: -300)
[20:49:25][C][tuya:054]:   Datapoint 101: raw (value: 01.01.01.01.01.01.00 (7))
[20:49:25][C][tuya:058]:   Datapoint 8: int value (value: 197)
[20:49:25][C][tuya:058]:   Datapoint 105: int value (value: 350)
[20:49:25][C][tuya:058]:   Datapoint 106: int value (value: 210)
[20:49:25][C][tuya:058]:   Datapoint 102: int value (value: 578)
[20:49:25][C][tuya:058]:   Datapoint 103: int value (value: 918)
[20:49:25][C][tuya:058]:   Datapoint 104: int value (value: 400)
[20:49:25][C][tuya:058]:   Datapoint 107: int value (value: 976)
[20:49:25][C][tuya:058]:   Datapoint 108: int value (value: 2000)
[20:49:25][C][tuya:058]:   Datapoint 109: int value (value: 500)
[20:49:25][C][tuya:058]:   Datapoint 110: int value (value: 1950)
[20:49:25][C][tuya:058]:   Datapoint 111: int value (value: 4100)
[20:49:25][C][tuya:058]:   Datapoint 112: int value (value: 410)
[20:49:25][C][tuya:058]:   Datapoint 113: int value (value: 1137)
[20:49:25][C][tuya:058]:   Datapoint 114: int value (value: 5200)
[20:49:25][C][tuya:058]:   Datapoint 115: int value (value: 520)
[20:49:25][C][tuya:058]:   Datapoint 116: int value (value: 1000)
[20:49:25][C][tuya:058]:   Datapoint 117: int value (value: 1110)
[20:49:25][C][tuya:074]:   Product: '{"p":"5ok0wq7drlqyo2hb","v":"1.0.0","m":2}'
```

Cossid, 2024-04-20:
Added as `Tuya Generic YY-W9099 Water Quality Meter v2.0.2`

Here is the API data returned
[image]

```json
"schema": [{
        "trigger": "direct",
        "type": "obj",
        "mode": "ro",
        "property": {
            "min": -100,
            "max": 1100,
            "scale": 1,
            "step": 1,
            "type": "value"
        },
        "id": 8
    }, {
        "mode": "ro",
        "id": 101,
        "trigger": "direct",
        "type": "raw"
    }, {
        "trigger": "direct",
        "type": "obj",
        "mode": "ro",
        "property": {
            "min": 0,
            "max": 1500,
            "scale": 2,
            "step": 1,
            "type": "value"
        },
        "id": 102
    }, {
        "mode": "rw",
        "property": {
            "min": 0,
            "max": 1500,
            "scale": 2,
            "step": 1,
            "type": "value"
        },
        "id": 103,
        "type": "obj"
    }, {
        "mode": "rw",
        "property": {
            "min": 0,
            "max": 1500,
            "scale": 2,
            "step": 1,
            "type": "value"
        },
        "id": 104,
        "type": "obj"
    }, {
        "mode": "rw",
        "property": {
            "min": -100,
            "max": 1100,
            "scale": 1,
            "step": 1,
            "type": "value"
        },
        "id": 105,
        "type": "obj"
    }, {
        "mode": "rw",
[... 642 lines cut ...]
        "typeSpec": {
            "type": "value",
            "max": 100,
            "min": 0,
            "scale": 0,
            "step": 1,
            "unit": "%"
        }
    }, {
        "abilityId": 141,
        "accessMode": "rw",
        "code": "rh_warn_min",
        "description": "\u663e\u793a\u8303\u56f40%\u81f3100%",
        "name": "\u6e7f\u5ea6\u4f4e\u62a5\u8b66\u503c",
        "typeSpec": {
            "type": "value",
            "max": 100,
            "min": 0,
            "scale": 0,
            "step": 1,
            "unit": "%"
        }
    }
]
```

closed 2024-04-20 by Cossid

LiesOfOmission, 2024-04-20:
awesome, thanks :heart: 
[image]

joohann, 2025-01-11:
Thanks for sharing! Works here too <3