tuya-cloudcutter/tuya-cloudcutter #653: MAC overwritten

powersupplyf, 2024-04-23:
Hi there,
tried using your fine program to OTA these [https://www.elektroda.com/rtvforum/viewtopic.php?p=21023310#21023310](url)

Had problems unfortunately:
[https://www.elektroda.com/rtvforum/viewtopic.php?p=21057441#21057441](url)

This is some of the journey's text from Putty
[cloudcutter.30154-error01.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15083130/cloudcutter.30154-error01.zip)

What am I doing wrong? 
This is the second time I somehow managed to overwrite the MAC

Also, how can I change this
 > 1.0.6 - BK7231T / oem_bk7231s_rnd_switch
to the actual file needed (I think it was OpenBK7231T_UG_1.17.553.bin)

Thanks

Cossid, 2024-04-23:
You can't override the mac address with CloudCutter.  CloudCutter only ever writes to the OTA partition and KV Storage, both via Tuya's SDK calls.  It does not touch TLV Store where the MAC Address resides.  The AP name in the log you provided is proof it still had a MAC address after being CloudCut.

You can only modify the MAC address via serial flashing over the TLV Store partition (or with a firmware that allows for very custom block address writing).  The only way to properly fix it is to restore a full backup.  Short of that, Easy Flasher has a tool where you can write a custom mac address (again, requiring serial access afaik), but it leaves all the wifi calibration data zeroed out, which may or may not effect your module's wifi performance.

Whatever you're doing, it is outside the scope of CloudCutter.

closed 2024-04-23 by Cossid

powersupplyf, 2024-04-23:
addition, really strange, I now see (only) on the main page some MAC, but unsure if it is the original
[image]

Cossid, 2024-04-23:
Then the firmware you installed (or choices you made within it) is responsible.  CloudCutter is incapable of doing that.

powersupplyf, 2024-04-23:
One final request pleeeease. 
You have such a nice tool, it has just one cosemtic issue, the deep blue colour of some of the text makes it really impossible for me to see. I tried changing my monitor etc, could only guess sometimes what it is showing.
Would be really great!
Thank you again.

powersupplyf, 2024-04-23:
got it, will wait for elektroda forum.

Cossid, 2024-04-23:
> it has just one cosemtic issue, the deep blue colour of some of the text makes it really impossible for me to see. I tried changing my monitor etc, could only guess sometimes what it is showing.

That isn't something we have control over, that is Docker, which we make calls to, but can't format its output.  Nothing in our own scripts we control uses any color coding.

powersupplyf, 2024-04-23:
Bummer, poor guy, being teased (only by me?) with so many issues you have nothing to do with.
I didn't know, deeply sorry and really impressed by your professionalism of keeping it calm albeit your frustrations.

Cossid, 2024-04-23:
I stand corrected, there is also color in the scroller for device/model/firmware choosing.  They aren't using colors we specified, I'll see if that is something we can adjust.

powersupplyf, 2024-04-23:
Really appreciate this!
Would be wonderful.

Cossid, 2024-04-23:
It doesn't look like we get a lot of customization there, so I made it as neutral as possible while being clear.  I changed the pointer character and left the text at the default color, but underlined it instead.

powersupplyf, 2024-04-24:
OMG thats great! 
Will try out later tonight with next 30154 OTA candidate, will report back.
Thanks