tuya-cloudcutter/tuya-cloudcutter #654: Moes Smart Curtain Switch

starbrights, 2024-04-25:
With cloudcutter I couldn't open the device containing a CB3S (BKN7231N) with Firmware 1.1.0
That is way I created a dump (by lightleak):
[dump_2024-04-25_11-34-04_lightleak.tar.gz](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15111745/dump_2024-04-25_11-34-04_lightleak.tar.gz)

I hope that can be used to create a profile for cloudcutter.

PS:
Would be great to add to the readme a manual how to flash the existion prebuild binaries for the needed CustomAP.
https://github.com/tuya-cloudcutter/lightleak/releases/tag/v2024.2.9
Without really nowing what I do:
esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB -ff 80m -fm dio 0x00000 lightleak-custom-ap_v2024.2.9_esp8266.bin
.. that makes it working.

Thanks to all contributors of this project!

Cossid, 2024-04-25:
Which model is this, do you have a product link?

starbrights, 2024-04-25:
https://moeshouse.com/products/eu-wifi-touch-curtain-switch

Model: WS-Y-EUTC-WH-MS, Wifi 2.4GHz
Board inprints are EU_CBL_SV, T=1.6MM, 20220409
20230318

Cossid, 2024-04-25:
Added as `MOES WS-Y-EUTC-WH-MS Curtain Switch v1.1.0`

closed 2024-04-25 by Cossid

starbrights, 2024-04-25:
Thank you so much. 
I tested it, but I have some problem (tried several times):  cloudcutter stops at here:
```
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlp109s0
Configuration file: /dev/stdin
Using interface wlp109s0 with hwaddr 00:c2:c6:e2:c0:e8 and ssid "cloudcutterflash"
wlp109s0: interface state UNINITIALIZED->ENABLED
wlp109s0: AP-ENABLED
``` 
I could see that my laptop (running Linux Mint) as created that AP and I can even connect to it from my phone with "abcdabcd".
The device (MOES) stops blinking but nothing happens any further.

Cossid, 2024-04-25:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do

starbrights, 2024-04-25:
More precicly:
```
================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlp109s0
Apr 25 16:01:33 dnsmasq[16]: started, version 2.85 cachesize 150
Apr 25 16:01:33 dnsmasq[16]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Apr 25 16:01:33 dnsmasq-dhcp[16]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Apr 25 16:01:33 dnsmasq-dhcp[16]: DHCP, sockets bound exclusively to interface wlp109s0
Apr 25 16:01:33 dnsmasq[16]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlp109s0 with hwaddr 00:c2:c6:e2:c0:e8 and ssid "cloudcutterflash"
wlp109s0: interface state UNINITIALIZED->ENABLED
wlp109s0: AP-ENABLED 
[16:01:34.565203 MQTT] Connected
```
Ok, just seen your comment. Will try that.

starbrights, 2024-04-25:
Well, step 1 and 2 is clear.
But in app it only allow that network I am in. So the description is wrong/misleading. First I need to connect with the app to that Wifi (cloudcutterflash). The problem is, I don't get an IP address and hence no connection in phone. So trying to add the device fails, because I am not in the network.
Anything wrong with my laptop regarding DHCP server?

Cossid, 2024-04-25:
You can manually connect your phone to the cloudcutterflash AP when at the screen to enter wifi credentials, and it should allow you to change at that screen, but yes, you need to be connected to a regular network before that point to load the app.

starbrights, 2024-04-25:
A regular network isn't a problem. But as soon as you want to change the SSID to cloudcutterflash you will end up in phone's wifi setting and it needs to connect to that SSID (which fails likely because I don't get an IP).

And If I change SSID to cloudcutterflash and return to password and confirming this I get the message: Phone is not logged in that network (which is true). It tries to connect but fails for some reason.  
I try to workaround that and give my phone instead of DHCP an static address 10.42.42.88. 
So I do have a chance to set AP-Mode in app, connect to A-xxx network and try to reset device - but with no success.

One question for understanding: 
In Cloudcutter: After apply the exploit cloudcutter it ask for power-cycle of device: Doesn't that make the previous exploit gone?!?

Cossid, 2024-04-25:
You can re-run the exploit as many times as you like before 3rd party firmware is restored.  The details provided by the most recent cut will be correct.

If you can't connect to the AP, then yes, you have some sort of issue, but I can't say what, nobody has reported the AP not working correctly before, it's always been issues with the target device.  Make sure the machine you are running on does not have any other 10.42.x.x network.  Running with `-v` will give more debugging information, possibly from dnsmasq/hostap.

If you can't connect to `cloudcutterflash`, you may have to run cloudcutter from a different device.

starbrights, 2024-04-27:
I tried a different (unfortunatly even older) laptop with no success:
I get messages:
Unbekannter Parameter: yes
.Unbekannter Parameter: --rescan
(but that seems to be no problem) and finally:

```
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlp2s0
Apr 27 08:49:50 dnsmasq[14]: started, version 2.85 cachesize 150
Apr 27 08:49:50 dnsmasq[14]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Apr 27 08:49:50 dnsmasq-dhcp[14]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Apr 27 08:49:50 dnsmasq-dhcp[14]: DHCP, sockets bound exclusively to interface wlp2s0
Apr 27 08:49:50 dnsmasq[14]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlp2s0: interface state UNINITIALIZED->DISABLED
wlp2s0: AP-DISABLED 
wlp2s0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlp2s0 wasn't started
Oh no, something went wrong with updating firmware! Try again I guess...
``` 
Do you think an update might help? Is it a matter of outdated software or speed or hardware(wifi adapter) capabilities?

Cossid, 2024-04-28:
There is a `-n` parameter to avoid rescan for older versions, but I have no idea what that hostapd error means or how to resolve it.