tuya-cloudcutter/tuya-cloudcutter #655: Tuya BHT-002-GALW (BHT-002GA) Water/Gas heating thermostat

icebrian, 2024-04-27:
Hello, 

I've been a long time user of the BHT-002GA thermostat because I managed to reflash OTA with fashberg/WThermostatBeca. Recently, I had one malfunction, so bought another, but this one was no longer possible to flash OTA and even soldering directly to chip because it was a diferent one :(.

I so decided to buy a Tuya BHT-002GA, hoping I could use tuya-cloudcutter, but alas, have also been unable to :(. I run the tool, select the device, its says exploit is applied, but when connecting to the device Wifi, I get a exploit not successull message.

Might this be because the only available device is the "Tuya Generic BHT-002GB WIFI Thermostat", so the GB version and not the GA version? (which by the way is the difference between Water/Gas heating for the GA version versus Electrical floor heating for the GB version).

Has anyone managed to flash a BHT-002GA? Or anyone have any tips?

Thanks,

Cossid, 2024-04-28:
What firmware version does your device have?

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

icebrian, 2024-04-29:
> What firmware version does your device have?
> 
> https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

Hi, 

Main Module: v2.5.2 
MCU Module: v1.0.0

Thanks,

Cossid, 2024-04-29:
We haven't seen any Beken firmware with a version anywhere near that.  It is possible you might have an ESP or Realtek module.  If it is ESP, you can try tuya-convert instead.

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

icebrian, 2024-04-29:
> We haven't seen any Beken firmware with a version anywhere near that. It is possible you might have an ESP or Realtek module. If it is ESP, you can try tuya-convert instead.
> 
> https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out

I got MAC via "Device Information" on the Smart Life app (hope that is the same as placing it in AP Mode and using Wifi analyzer) and MAC reads as belonging to Tuya Smart Inc with OUI 18:DE:50

Cossid, 2024-04-29:
You'll need to verify which module is inside your device then, as this device likely has an MCU in it, and the highest known beken firmware version for MCU is 2.1.6.  If it is Beken, we'd need a full firmware dump to possibly support it, provided it's not patched.

There is also the possibility it is Realtek as well, in which case your options will be more limited.

icebrian, 2024-04-29:
Module is a WBR3, additionally on the board itself I have ME86H-WIFIMBXIN-V1.5 (if that helps).

How do I go about with the firmware dump? Will be glad to provide it.

Cossid, 2024-04-29:
WBR3 is Realtek RTL8720CF and not currently supported by CloudCutter.  There is limited support from LibreTiny/ESPHome via serial flashing, but many unsupported things still like OTA, and wifi is still considered unstable.

closed 2024-04-29 by Cossid

icebrian, 2025-02-04:
Hi @maxdd , can you point me to more info on the custom firmware?

maxdd, 2025-02-04:
This is a good reading

https://www.elektroda.com/rtvforum/topic4097185.html

The WBR3 is a RTL87X0C chip and there is an openbeken firmware that can be used https://github.com/openshwprojects/OpenBK7231T_App/releases

I'm planning to buy a BHT-006 and start my "flashing" activity once i can't stand anymore the original product but if i know there is a way without desolder the uC i would be even more convinced to do so

Cossid, 2025-02-04:
There has been no progress on supporting Realtek on the CloudCutter platform.  It's not simply a matter of having firmware dumps, it's a matter of knowing how to direct exploit control at the assembly level, which is not currently known, and there is no timeframe.

Cossid, 2025-12-17:
> Is v3.2.0 supported? Can I flash my thermostat? Model code is ME82

v3.2.0 is not currently known, so we would need either a dump or a serial boot log to make any guesses.