tuya-cloudcutter/tuya-cloudcutter #656: nl80211: Could not configure driver mode

starbrights, 2024-04-28:
Hi, seems I am failing also with my 2nd laptop tu run script.   What is reason for this problem? Anything I can solve?
Thank you.

```
================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlp2s0
Apr 28 07:18:24 dnsmasq[15]: started, version 2.85 cachesize 150
Apr 28 07:18:24 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Apr 28 07:18:24 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Apr 28 07:18:24 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp2s0
Apr 28 07:18:24 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlp2s0: interface state UNINITIALIZED->DISABLED
wlp2s0: AP-DISABLED 
wlp2s0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlp2s0 wasn't started
Oh no, something went wrong with updating firmware! Try again I guess...
```

Running on a 2404 Ubuntu Mate.

Cossid, 2024-04-28:
I can't help with this error, I have no idea why hostapd won't run with your wifi card.  That is beyond the support scope for Tuya-CloudCutter.

closed 2024-04-28 by Cossid

starbrights, 2024-04-29:
Running script on another PC helps. Note: Have different phones/PCs to test!