tuya-cloudcutter/tuya-cloudcutter #657: Deta DET902HA Series 2 Downlight - WB2L

Br3no94, 2024-04-29:
Factory device dump via ltchiptool. 

WB2L module, device info from ltchiptool;
chip type: BK7231T
Bootloader Version: 1.0.6
Bootloader Message: BK7231S_1.0.6
Protocol Type: BASIC_TUYA

[ltchiptool_HT-DET902HA_S2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15144982/ltchiptool_HT-DET902HA_S2.zip)

Br3no94, 2024-04-29:
Storage area JSON added from ltchiptool
[DET902HA_S2.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15145035/DET902HA_S2.json)

Br3no94, 2024-04-29:
I was able to cloud cut the device to ESPHome custom firmware (not kickstarter) using firmware version 1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty
this loads device slug: hykolity-rgbct-downlight
profile: oem-bk7231s-ceiling-light-ty-1.3.2-sdk-1.0.2-40.00
so a profile may not need setting up? unsure what the go is here.

Cossid, 2024-04-29:
Added as `DETA DET902HA Series 2 Downlight v1.3.2`

closed 2024-04-29 by Cossid